opentx/opentx #3481: Slowdown goes from -80 to -80 over 0 when custom curve used

janekx, 2016-05-10:
Tested on 2.1.18 and older have same bug. 
When in mixes are used slowdown and curves like CH2 together we got other behaviour then when is used function.
There is no respected slowndown set. 

[image]

and curve 
[image]

If is curve change to function like X>0 all works well 
As well the behaviour in switching this on is other 

when is used curve the value skip in place when switch is activated 
when is used function the value goes in place respecting slowndown when switch is activated

Interesting is to use throtle like source and try to switch SG up and down and see how it react when curve is used and throtle is down. It generate bad signals from -100 to 0 but should not.

projectkk2glider, 2016-05-11:
Slow and mixer switch are not meant to be used together. Use another higher channel to prepare the value ( no slow here) and then use that channel as a source for CH2 with the slow.

janekx, 2016-05-12:
Thanks,  interesting that this happened only when is used slowdown down and up works well.

closed 2016-05-27 by projectkk2glider