opentx/opentx #3484: Taranis as Slave with Dx5e as Master - Servo twitching and Throttle goes up and sown

rguptaca, 2016-05-11:
See video.
https://youtu.be/qAxRKkfjBKg
Servo twitching and throttle up then down every 6 to 30 seconds.
Dx5e as master. 
Spektrum trainer cable (the official expensive mono to mono cable).
Apprentice S 15e Rx.
FrSky Taranis Plus Tx.
OpenTx 2.1.8 2016-04-14 EEPR 217
PPM Frame default settings of 22.5ms 300u -

I and able to control the plane with the Dx5e when the Trainer switch is Down.
I am able to control the plane with the Taranis when the Trainer switch on the Dx5e is Up but there is servo twitching and the motor starts and stops on its own.

We have been using this setup with Spektrum transmitters in place of the Taranis and it works. We only have this problem when trying to use the Taranis.

As far as using a stereo cable goes the Taranis has the ground and middle/second conductor shorted (I checked with a multimeter) so it signalling on the electrical/physical layer is probably not be the problem.

projectkk2glider, 2016-05-11:
Did it work in any previous OpenTX version?  Did you try to change the PPM frame settings and number of channels sent? I presume channel values are stable on Taranis ( it is not visible in the video).

rguptaca, 2016-05-11:
I also tried using a Taranis (not plus) with the same problem.
OpenTx 2.1.3 2015-10-01 EEPR 217
I used default values. 
Channel values on Taranis are stable when viewed using the channel monitor.
I hesitatte to change values because I don't want't the plane going to full throttle unexpectedly. I don't have someone with me at the moment to hold the plane.

projectkk2glider, 2016-05-11:
Remove the prop! Then try changing the polarity first.

projectkk2glider, 2016-05-11:
I am sure you will get better answers on forums, there is bound to be sone user with a setup similar to yours.

rguptaca, 2016-05-12:
Based on a response to my forum post and this thread http://www.rcgroups.com/forums/showthread.php?t=2416188
I changed 
PPM Frame to 22.0 ms 400u -
I also changed OUTPUTS so that the min max ranged from 1100 - 1900 ms with centre on 1500 ms (I had it at 1520 ms). I may change the min max range back to the defaults to see if it still works but for now I've spent enough time on this.
So, would it be worth making a feature request to have a menu to be able to select PPM Frame settings
- Spektrum
- Futaba
- JR
- Graupnier
- Custom
- Other popular Tx in use

closed 2016-05-12 by rguptaca

kilrah, 2016-05-12:
It's the first time I hear about settings needing to be changed so it doesn't seem worth having 2 entries for "Some version of Spektrum DX5e" and "Everything else"...

rguptaca, 2016-05-12:
https://github.com/opentx/opentx/issues/140
In this thread it was discovered that Spektrum has a different implementation. 
I wondered myself why there were not more reports of this problem. Anyway, now that Ive done a few posts and a video maybe the next person will have an easier time.

mwigman, 2017-06-04:
I am having the exact same problem using Spektrum DX4e as master and Taranis Plus as slave (Taranis bought spring 2017 with stock version OpenTX). Motor is pulsing every 1-2 seconds in low power part of the throttle range, at high power it works fine. I have not tried changing ppm settings on the Taranis yet, but it also occured to me that there is a mismatch between Taranis and Spektrum ppm convention, I will start changing the settings lateron this week. I guess this problem is mentioned nowhere because mostly the Taranis is used as master and not as slave, especially combined with cheap transmitters as master. In my case I try to help a fellow beginning pilot with Spektrum setup (TX and RX) using my Taranis as slave. In that way he can fly with my Taranis as buddybox, not having to replace his receiver.

kilrah, 2017-06-04:
It's most likely a signal level or coupling problem, which cannot be controlled by OpenTX but would need a hardware solution.
AFAIK Spektrum uses a low level capacitively coupled signal.

mwigman, 2017-06-04:
As I do not know for sure if @rguptaca solved the issue by changing the ppm settings I will try myself first. Does it make sense to conclude it is a signal/coupling issue if the connection does work the other way around, with Spektrum as slave? Many people use it successfully as I have read.

kilrah, 2017-06-04:
Yes. The Taranis' input circuit was made to **accept** pretty much anything, but that doesn't mean every other radio will accept what it puts out.