opentx/opentx #3488: Special function suggestions

elecpower, 2016-05-13:
Allow Enable on/off for all functions - saves changing/removing the trigger. Helpful when testing, temporarily disable or you include a standard set of SPs for all your models but some do not apply to this model as currently configured. (I think this was raised previously but cannot find the issue).

Change default Action from "Override CH1" to "---". Logically Switches use this principle. IMO there should be no valid actions and force the user to select.

elecpower, 2017-11-16:
Channel names added in 2.2

AlgebraicMirror, 2018-04-02:
I'm not sure if I should create a new suggestion, or use this existing one, since my suggestion is related to special functions. If I should put it elsewhere please let me know.

On the special functions page, the channel override function doesn't allow sufficient granularity to be able to match the expectations of certain Spektrum receivers/FCs. For example, when using the override function, you can pick values like -107 or -108, but not -107.5. In other screens, such as the output page, you can set limits like -78.2, and the extra decimal place allows for very precise tuning down to the individual microsecond.

Here is an example of why this same level of granularity is needed for function overrides. Horizon Hobby's X-Vert model has an extremely precise requirement for a throttle cutoff switch (and throttle cutoff is 100% required for landing and for safety in the this model, since once armed the props operate in a Betaflight-like "air mode" and will not stop spinning even if the throttle gimbal is completely lowered). You must send the X-Vert a very exact throttle value (equivalent to Spektrum -130), and it must be within +/-1us. This is typically done by assigning a channel 1 override to a switch. However, there is a difference of several microseconds between the override values of -107 and -108 (which are the Frsky X7 approximate equivalent to Spektrum -130), and I found that for my particular transmitter/external module neither of those was close enough to trigger the throttle cut function on the X-Vert flight controller. However, if I modified Channel 1 on the output page so that PPM center was 0.2 instead of 0.0, and then set my throttle cut to -107, that extra .2 variance was enough to get it into range, and the throttle cut functionality started working.

But this is really an undesirable hack, because the PPM center point did not need modification and should have been able to stay at 0.0. It was the channel override that need modification, but OpenTX did not allow enough granularity to do so, since you can't have decimal points on that page. If manufacturers like Spektrum are going to manufacture products with such tight tolerances that you can't be off by more than 1us, then OpenTX will need to add granularity in places like Channel Override in order to match them.

Here is the RCGroups thread where this example with the X-Vert and the Frsky QX7 is discussed in more detail.

https://www.rcgroups.com/forums/showpost.php?p=39432416&postcount=7508