opentx/opentx #3491: Model settings user defined paging sequence

elecpower, 2016-05-13:
I'm raising this as a separate issue as there have been lots of comment (recently and in a quick scan of the issues way back) about too many screens to page through. This argument is used to push back on any suggestion of another screen. However as more radios hit the market the amount of functionality is likely to keep increasing eg Horis. OpenTX already supports a vast array of RXs with basic through to advanced functionality. So more and more fields and screens will likely be necessary in the future. On the other hand those with older radios will not use or rarely use the newer features. 

So why not allow users to configure the paging sequence and even hide screens they never or rarely use especially in the field. Not suggesting by hide to remove from firmware like the heli option though over time this may make sense for some. User choice is already implemented for default channel order assignment. OpenTX prides itself on being flexible compared to those proprietary products.

For myself as reasonably new to RC and OpenTX, I do most of my more complex set up in Companion and tweak via Radio in the field. Using the Simulator to test in the safety of the virtual world what prove to be dumb ideas especially for LSs or SFs is a model and dollar saver. Fantasic feature!!!

JimMchugh, 2016-06-01:
+1
I never use Flight Modes or Curves (use normal Expo) for example. Would love to hide those...

Oldgibber, 2016-06-10:
The downside of this is that it would be a nightmare for instructional videos or for documentation, or down at the field when one tries to help another user.

elecpower, 2016-06-11:
@Oldgibber fair point but there are already numerous firmware options and general settings which affect the look, feel and operation. That is a beauty and the curse of OpenTX. If people in general do not like this then they should buy those other brands we do not mention here...

lshems, 2016-06-26:
Well, there are some that write a user interface on top of OpenTX :| 

That works perfectly with the beautiful LUA interface we have. Just search for 'Taranis Apps' on google.

projectkk2glider, 2016-06-27:
@lshems please refrain from using this site to promote your **commercial** solutions.

lshems, 2016-06-28:
I'm not promoting any commercial solution. . 

I think I'm merely pointing out to solutions available on the net to tackle the limitations or requirement subject of this thread, to be found by googling. Whether commercial or not, that is not relevant in my opinion when used as an example on how things can be done.

How often did someone refer to some way things have been done in the Multiplex transmitters, to propose similar solutions to be incorporated in OpenTX? 
I could as well have given the Mikado transmitter as example on how to use scripts to create custom user interfaces for tedious programming actions. They have apps as well.  
Jeti is going to introduce LUA on their next models, for similar reasons.
And the Horus and Taranis, mentioned a million times because they are the transmitters for which OpenTX is developed, do you regard those references as "commercial" promotion as well?

There are numerous commercial and non-commercial examples that could point at how to use Scripts to create custom user interfaces. I choose to hint at apps for the Taranis. That's where OpenTX is used, no?

I'm proud of the solutions I've created, as you are probably on yours, and referring to the approach taken as example on how to do things is not intended as "commercial promotion". If the approach is taken over by others in the form of "free" solutions, then that's fine, and everyone is free to do so. I do also share code and best practices for using LUA, on forums and in private chats. That's what this free world allows us to do, choosing the means to achieve what you want.

But since the OpenTX community, or some within at least, seems to have developed a rash on anything that could be interpreted as "commercial", I'll try to stop scratching it before it starts bleeding.

Try to remain fair, and don't jump at everything that could be interpreted as commercial, but is not intended as such.

kilrah, 2016-06-28:
> I'm not promoting any commercial solution.
I think I'm merely pointing out to solutions available on the net

You are, because unlike what you suggest there's only one and it's yours.

> But since the OpenTX community, or some within at least, seems to have developed a rash on anything that could be interpreted as "commercial", I'll try to stop scratching it before it starts bleeding.

We have nothing against commercial solutions if done right, but the way you've gone about it is simply not fair. 

There are 2 ways people who work on the project will do so:

- There is something they like and think will benefit the project, e.g. a cool new platform, a new feature they want or read about and think is interesting and worthy to have etc. They will be happy to donate their time to work on it when they please because they have fun with it.
- Someone (a manufacturer, a reseller or other commercial venture) wants something that will benefiit them but that the team has no particular interest in or which does not really go in the project's intended direction, and they want it done now. In this case they will offer to pay the developer(s) who will do what is now a job rather than fun, or donate some of their proceeds.

Now in the 2nd case if the requester is going to offer whatever they want for free or even better open source it then it's for the good of all and we will likely do the thing for free. They are going to donate time for the good of the community thanks to our platform, so we're happy to help them do it. 

In your case we spent a significant amount of time in the past year discussing issues and implementing things that are not going towards our goals for the project and are of use only to you (again, unlike what you suggest with "there are **some** that write a user interface on top of OpenTX" there's actually only one and it's you). That would have been OK if you offered your scripts for free for the reason mentioned above, but by deciding to go commercial you decide your time is worth money - in that case ours we spent for you is too. 

People donating their time for non commercial stuff are happy to all work together for free for the good of all. Once you go commercial and involve money you start doing business, and it's not the same thing anymore. It's a perfectly valid thing to do, it's your choice, but if you make it you should accept everything that goes with it, like paying for your advertising and for the work others do for you. That would be fair and that's what our other partners do.

Unfortunately you're doing the opposite, sneakily post about your commercial stuff in forums everytime you get a chance just like you did here in a completely unrelated discussion (that's spam, and I know that you were repeatedly reported for it), and try to get away from recognising the work that was done for you by suggesting others use it too...
It's that arrogant attitude that's also carried over on your website that gets you flamed in the community.

lshems, 2016-06-29:
As said, I won't scratch again. 

But I don't agree with your analysis and allegations.

Thanks

lshems, 2016-06-29:
https://www.youtube.com/watch?v=yRo_L1RiGWw

schwabe, 2016-06-29:
@lshems linking to YouTube like that, also makes it hard for people like me to stay neutral and positive. Try to think about that.

lshems, 2016-06-29:
I don't think that you have understood the message of that video. 

It is telling that people cannot value things that have no cost. But it is acknowleging that things that have no cost (can) have great value, but it is just not recognised by the people.

I posted it because in a way it is meant as a compliment to free open source software, but at the same time points at the weakness of it being for free.

Please try to take the bias away, and look at things as they are. It is not meant negative, nor personal. in contrast to some other posts you can find here and there.

Best to all, and long live open source.

schwabe, 2016-06-29:
@lshems I understood it as message that you sell yourself instead of giving it for free because people will otherwise not appriciate it.

jtaylor2, 2016-06-29:
@lshems I do not appreciate being spammed with YouTube video links in my bugmail.  If this was my repository I would block you.  I remained silent when you spammed RCGroups  with promotions for your commercial application site.  And the promotions would not have bothered me near as much if you had referred to it as your site instead of referring to it in the third person as if it belonged to somebody else. But spamming everybody here because they don't see things your way is a bit too childish to ignore.

Jim

kilrah, 2016-06-29:
> it is acknowleging that things that have no cost (can) have great value, but it is just not recognised by the people.

That is completely wrong, we wouldn't be doing this if people didn't recognise its value, it's actually the very thing that makes us do it. There isn't any bigger proof of recognition than the fact that dozens of thousands of people are using something that was not initially meant for them. If they do the above-average effort needed to use it, it's obvious that's it's highly valuable to them.

You're the only one I know who either doesn't recognise its value or tries to downrate it and make people believe it's  limited until your commercial stuff is added on top... which precisely contradicts itself since if the platform had little value you wouldn't have made something on top of it.

Just understand that your message is inconsistent, and most of the things you post here are seen pretty much in the opposite way it appears to you by everybody else, so if you want to participate please respect that and refrain from trying to impose your point of view.

Mow06, 2016-06-30:
Kilrah, your patience amazes me!!!

lshems, 2016-06-30:
Well, people often get misunderstood.

I'm not downgrading OpenTX, or at least that has never been my intention.

I am still convinced that OpenTX, and the Taranis more specifically, is not useable for a lot of people, for a lot of reasons, and I experience that every day. 

Your standard reaction has always been that people should either learn OpenTX, or they are not your "audience".

Well, you do not pick your own audience, and as I have pointed out, I think that that poses some responsibility on you. You have made clear that you don't share that opinion, and don't see that as a problem, and I respect your opinion, although I do not agree.

It is this reaction and position which pushed me to go ahead and create something which would possibly lever the use of OpenTX, and attract even more people into using OpenTX, by reducing it's complexity for achieving simple things. And I don't believe OpenSource is the optimal way to do that, for several reasons.

I don't think any of you really understood the message in the youtube video. My "paying" customers are pictured as "poor" inconscience people. In the same anology, the users of OpenTX that don't need the "paid improvements" because they want t o learn and master OpenTX are the conscious ones. 

This underlines my point that we need both: OpenTX for the serious hobbyist, and some Apps to make it useable for the Bind And Fly amateurs that don't want to understand all that they are using. 

But OK, I am arrogant, I downgrade OpenTX, and I have the worst intentions with OpenTX. Than that is clear at least, and nobody can misunderstand your opinion there.

JimMchugh, 2016-06-30:
@Ishems: Shame...

Many highly skilled and dedicated guys here, producing quality software for the good of the flying community.

I don't mind OpenTX on Taranis or Horus (I have both) being the slightly "techie" operating system, but with the highest level of flexibility and customisation as *any* radio system.

My fellow club pilots are constantly amazed at what we can make OpenTX do, and then talk about buying *another* radio from another manufacturer because it has a certain mix or function.

Crazy!

kilrah, 2016-06-30:
> Well, you do not pick your own audience

No, but we pick the direction we want to go in. The audience either follows or not. 

> It is this reaction and position which pushed me to go ahead and create something which would possibly lever the use of OpenTX, and attract even more people into using OpenTX

OpenTX with a layer on top of it is not OpenTX anymore. That is not what we're trying to do.

I fully understand what you're after, but as I already mentioned making a layer on top of OpenTX is the worst possible way to go about it. 
Just look at the instructions on your website... getting your scripts to run involves pages of steps, which are filled with most of the things that people typically consider complicated and complain about. 
Models being configured by your scripts the entire OpenTX interface becomes redundant, confusing, and prone to causing peple to mess things up (tweaking things after the script has run and making sure you're not futzing things involves understanding from the user that if they had they wouldn't need or want your scripts).

If you really wanted to make it more accessible you'd do like FrSky for the Horus and make a separate firmware with a completely different interface. It could be open source, it could even be based on OpenTX "under the hood" for the calculations, but the entire top layer should be replaced with something appropriate for that completely different goal.

lshems, 2016-06-30:
That's your point of view. I have a different one, supported by the users opinions. It is perfectly feasible to do a basic setup with a custom user interface, and have the tech savy people adopt or add settings.

I have another business model. If the Apps don't please my audience, I wouldn't have an audience.

If you don't please people with OpenTX, they are just not "your audience".

tomkfly, 2016-06-30:
I for one am quite happy with OpenTx and I'm sure there are many thousands of others that agree with me. I find your posts on here are out of order, as are your advertising posts on RC Groups.

kilrah, 2016-06-30:
> If you don't please people with OpenTX, they are just not "your audience".

Exact, and I'd rather prefer that people who are not pleased by OpenTX the way we make it don't use it instead of using a kludge on top of it and have a result that is not up to our standards. 
An offering directed to them may be desirable, but should be "done right" and from my point of view that can only be done as a separate project.

> I have another business model.

So if you have a business model you confirm you are running a business, and as such the work that some of the project members have done to specifically accomodate your business because it is not in line with our goals for the project are a business service... as such you should expect to pay for it.
As explained earlier if your business model involved providing the result for free we'd do our part for free as well, but given you decide to charge for it then so will we.

Feel free to continue posting issues as you see fit, but understand that while some may fall in line with our goals and/or interest some of our developers and be implemented quickly some may not and will either linger in the issue tracker for an inderterminate amount of time or be dismissed completely. If you want them addressed in order to achieve your business' goals, feel free to do the coding work yourself and submit pull requests, or hire someone to do it whether an external dev or one from the OpenTX team. You can contact us privately to make offers or request quotes.

Hopefully you'll understand the reasons for some of your suggestions not being addressed, and the consequences of your business decisions.

lshems, 2016-07-02:
Well, it becomes clear that you dont like my approach on opentx. Well, with the risk of being regarded arrogant, thats not my problem.

I do fully agree on your point of view regarding payment for developing  functions that you don't see within scope of your main goal. There is nothing against that.

But I see no reason why my approach with apps is good or bad. If the apps cover all needs for someone, he can use them without using opentx. If he wants more, he can dive into opentx, without changing transmitter or epee, and add it on top.

I think that even if I would offer my products for free, I would get more or less the same reaction here, becaase i can imagine it is annoying to have someone promote, or point out, to solutions using opentx in a way you don't like.

Of course I'm also arrogant, especially on my site, so that's enough reason to jump on everything which is related to what you don't like. 

My conclusion remains the same, opentx community is having an itch.

And i've already said I'll try not to scratch it.

P.s. apart from being arrogant, I've never called names or made discussions personal. But I've already mentioned as well that I think discussion s and reactions here sometimes lack some respect. And we differ also on the definition of respect, so don't return that one making reference to me not having respect by doing what you all see as 'advertising'.

Last point for me, I'm thinking this whole discussion is useful in a way that positions become clear.
I certainly wasn't expecting such an attitude towards the apps solution I'm providing, free or paid.
Rather enlightening.

:)

Best to all, and keep up the good work. And don't let an arrogant guy ruin your fun.
The arrogant guy doesn't want that either.

mpjf01, 2016-07-02:
Can we talk about the OPs post now? There's a suggestion to allow the "hiding" of Flight Modes and Curves, It would be possible to hide the FM setup screen I guess, but what about all the other references to them? In mixes for example and elsewhere. Is not having to press the page button one more time really worth the effort involved in removing the option and references to it? And then someone will post an eepe with them in and the recipient user probably won't know what they even are. My preference is to leave the stuff there, that way new users become exposed to it, learn about it by osmosis and asking questions and the downside is really pretty minimal.

bsongis, 2016-07-02:
Yes it is possible to hide either FM setup screen, it also removes references to the FM in all screens. The same with CURVES. I just need to add this option in Companion, the firmware already has this feature implemented

kilrah, 2016-07-02:
As already said I am not favorable to it since many people rely on files posted on the web and will become confused when there are settings done in pages they've hidden.

I already see that problem happen at least once a week already with GVARs, the override function or simply people not even thinking to go look for a setting on a page they have displayed but don't think could be related so they don't go and see it.

To me it should not be possible to hide anything. The more info the better.

projectkk2glider, 2016-07-02:
I also strongly believe that hiding some pages will only create problems, both for users and for us.

griebd, 2016-07-04:
I tend to agree with @kilrah and @projectkk2glider, it can create a lot of confusion... but then, what about a more complex solution?! there could be a compile option (not the default one) to enable the hiding of screens/options! then, those who wants to use it would need to take the more "tech" path...

schwabe, 2016-07-04:
@griebd That makes it even worse. That means that is an obscure options that creates even more obscure problem.

The only way I see for this problem is that hiding pages is only allowed if they show only default values and automatically reverted if not showing default values anymore.

bsongis, 2016-07-04:
That's exactly what I proposed:)

3djc, 2016-07-04:
Hiding screen through compile would be quite dangerous when sharing models. Would the function be disabled at great risks ? Will it be there but invisible. Also, having everybody seeing the same screen (in the same order) helps a lot when helping someone over the phone...

kilrah, 2016-07-04:
What @griebd says is a compile option to _enable the hiding of pages_ that is then done in the menus - i.e. unless you check that option you can't hide pages, so users who haven't taken the step of selecting the option aren't at risk (although I've seen more than once people who clicked random options without checking what they'd do and were then lost, but that's another problem...)

> The only way I see for this problem is that hiding pages is only allowed if they show only default values and automatically reverted if not showing default values anymore.

That I'm fine with, didn't realise that's what @bsongis had in mind. When is the check done? At each model load, and if a non-default value is found the corresponding "hide page" general setting is cleared?

bsongis, 2016-07-04:
my last answer was sent after the other post, doesn't make sense now. and no it's not implemented this way

kilrah, 2016-07-04:
OK, the compile option to enable the "hide page X" choices then

closed 2019-11-15 by stale[bot]