opentx/opentx #3492: Verfy the actual delay of delay_01us() function

projectkk2glider, 2016-05-14:
Some test suggest that it is not right. Test environment in 2020807473d468fbf13db5798610d87d1c7e7c0e

projectkk2glider, 2016-05-14:
With commit https://github.com/opentx/opentx/commit/4fe6122141552ae4abd08868ba3cde5fb7f221c7

On Horus:
```
delay 1ms  : 0.119ms - 0.120ms
delay 2ms  : 0.239ms - 0.239ms
```

Looks very wrong, now I am not sure if it is the delay or the measurement problem -ie debug timers. I will try the same on Taranis....

projectkk2glider, 2016-05-14:
Similar *problem* on Taranis (tested with 87229ed4119840077c4e0364c812ab28dc14038b):
```
delay 1ms  : 0.167ms - 0.167ms
delay 2ms  : 0.334ms - 0.334ms
```
:fearful:

bsongis, 2016-05-14:
It seems we have a *10 factor!

projectkk2glider, 2016-05-14:
New method - same results (on Taranis):
```
delay 1ms  : 0.168ms - 0.168ms
delay 2ms  : 0.334ms - 0.334ms
```

projectkk2glider, 2016-05-14:
Something fishy going on, with the timer prescaller multiplied by 2, I get the same times:
```
delay 1ms  : 0.168ms - 0.168ms
delay 2ms  : 0.334ms - 0.334ms
>p tim 13
TIM13
 CR1    0x2
 CR2    0x0
 DIER   0x0
 SR     0x2
 EGR    0x0
 CCMR1  0x0
 CCMR2  0x0
 CNT    0x2724
 ARR    0xffff
 PSC    0xb
 CCER   0x0
 CCR1   0x2710
 CCR2   0x0
 CCR3   0x0
 CCR4   0x0
```

projectkk2glider, 2016-05-14:
Delay fixed, on Taranis:
```
delay 1ms  : 1.001ms - 1.001ms
delay 2ms  : 2.001ms - 2.001ms
```

and on Horus:
```
delay 1ms  : 0.953ms - 0.953ms
delay 2ms  : 1.905ms - 1.905ms
```

bsongis, 2016-05-14:
Cool, thanks for fighting it!

kilrah, 2016-05-14:
Cool!
Is it normal to be that off on Horus?

projectkk2glider, 2016-05-14:
Yes it is normal, the timer clock is 84Mhz, so ideal prescaler would be 8.4-1. I will make adjustment for delay_ms() function.

projectkk2glider, 2016-05-16:
@bsongis While the same problem exists in the `master` branch I am reluctant to make this change there. Do we agree and close this issue?

bsongis, 2016-05-16:
Yes, it would bring problems

commit 59653f9c (2016-05-14): Fixes #3492: delay_01us() and delay_ms() functions were generating wrong delays (6 times shorter on Taranis, 8 times on X9E and Horus). Important delays adjusted, function names and code unified. (#3496)
PR #3497 (2016-05-15): Fixes #3492: better implementation of delay_01us() using Cortex syste…
…m cycle counter

Closes #3492

I think this is a better solution than #3496. It uses Cortex system cycle counter register. Some references:
* http://www.carminenoviello.com/2015/09/04/precisely-measure-microseconds-stm32/
* https://kbiva.wordpress.com/2013/03/25/microsecond-delay-function-for-stm32/
* https://github.com/spark/firmware/blob/9a99ba61c7b6399ccdf8a1238d05803beb7bd1ee/hal/src/stm32f2xx/delay_hal_stm32f2xx.c#L32

Tested on Taranis and Horus, looks good.

To test the actual delay use branch https://github.com/ope [... cut]
commit a0feed7f (2016-05-15): Fixes #3492: better implementation of delay_01us() using Cortex system cycle counter (#3497)
closed 2016-05-16 by bsongis