opentx/opentx #3507: [Telemetry simulator] RPM seems to be constrained to maximum 65535

raphaelcoeffic, 2016-05-18:
This is a bit too low for turbines. Propose to raise this maximum value to 262143 (0x3FFFF). Pull request follows.

projectkk2glider, 2016-05-19:
We have to check what is the official range.

kilrah, 2016-05-19:
0-60'000 per the doc

3djc, 2016-05-19:
The official frsky rpm sensors(http://www.frsky-rc.com/product/pro.php?pro_id=136) is rated for 300k rpm, not sure how they are sending that tho (RPM range (for 2 poles brushless motor): 1000rpm to 300000rpm)). (Pole count is managed at sensor level, not through blades on the radio)

raphaelcoeffic, 2016-05-19:
The point is, there is no issue neither transmitting nor receiving such a value in the firmware. This issue only concerns the telemetry simulator in companion. So basically, I don't see why opentx should understand & display telemetry values in an inconsistent manner between the TX and the simulator. The values are logged properly in the TX, but it limits it when replaying the log in the simulator.

raphaelcoeffic, 2016-05-19:
@3djc: does that means that I should raise even higher? say 0x5FFFF ?

projectkk2glider, 2016-05-20:
The SPORT spec says:
 * RPM type is uint32_t
 * RPM range is 0-60000

The spec is therefore in a bit of contradiction (which is not unexpected) and the spec from actual RPM sensor proves that the max value is bigger than 60000.

The OpenTX code treats incoming uint32_t value as int32_t therefore any value in range from 0 to 0x7FFFFFFF (2147483647) will be processed correctly. I recommend you alter the pull #3508 to 0x7FFFFFFF.

raphaelcoeffic, 2016-05-23:
@projectkk2glider changes made and moved to pull request #3506 as you requested.

projectkk2glider, 2016-05-23:
Done in #3506

commit 490b5531 (2016-05-23): Closes #3507: raise max RPM in telemetry simulator to 2147483647 (0x7FFFFFFF)
commit 0f1259d4 (2016-05-23): Closes #3505: add fuel quantity sensor in telemetry simulator (#3506)

* Closes #3507: raise max RPM in telemetry simulator to 2147483647 (0x7FFFFFFF)
commit 06fce510 (2016-05-23): Closes #3505: add fuel quantity sensor in telemetry simulator (#3506) (ported from master)

* Closes #3507: raise max RPM in telemetry simulator to 2147483647 (0x7FFFFFFF)
closed 2016-05-23 by projectkk2glider