opentx/opentx #3510: Heli setup not properly backuped

blairfancy, 2016-05-19:
Hi,
I ve noticed that on companion 2.1.8 heli setup is not properly saved/restored especialy inversion on dir/ail/pitch.

Regards
Thierry

projectkk2glider, 2016-05-20:
Please describe the issue in more detail, we are really illiterate about helis :wink:  

I tried setting all values in Heli tab, saving to eepe and the reopening the file. All values were there. Do you see a problem when transferring from/to radio?

blairfancy, 2016-05-20:
Ok here are the step to reproduce issue:
-first setup model and in heli setup invert channel Pitch
-with companion load models from radio
--finally check heli setup in companion you can see that pitch channel is no more inverted (no check in the box)

hope it help...

schwabe, 2016-05-20:
@blairfancy OpenTX multiple ways to invert a channels. In which screen did you invert the pitch channel? (Mixers/Outputs/Heli...etc)

It would also help if you can make a backup of your eeprom from the remote and attach the eepe file so we can load it ourselves in Companion

blairfancy, 2016-05-20:
as explained invert channel in "Heli Setup" screen
ok i ll try to backup model with radio itself

schwabe, 2016-05-20:
That screen has no invert option?! (I am also not a heli user)

[image]

blairfancy, 2016-05-20:
In the radio there are 
[image]

[fbl100.txt](https://github.com/opentx/opentx/files/275359/fbl100.txt)

griebd, 2016-05-20:
my guess is he is importing from an old version... this setup has changed!
there is a upgrade note saying about it!!! before we had the invert option, now we have weight! there is no automatic conversion though...
@blairfancy, is my guess correct? are you importing a eeprom from an old version of openTx in a newer one?

blairfancy, 2016-05-20:
@griebd, no and bug can be reproduced with companion only (radio simulator). bug is only on model setup of companion on radio no issue.

projectkk2glider, 2016-05-20:
I see now, the inversion parameter is present on platforms that don't support *virtual inputs* like 9X and 9XR, etc..

schwabe, 2016-05-20:
Yes. I Taranis encodes the paramter in the weight, negative weight is inversion. When reading the eeprom for 9x we do not store the value in the SwashRingData (opentxeeprom.cpp function, class HeliField). Writing back that config should work. But that flag cannot not be modified via companion

schwabe, 2016-05-20:
@projectkk2glider are you already coding a fix? Otherwise I would take this.

commit 43b0aeb2 (2016-05-20): Fixes #3510: Fix heli inversion not correctly shown/saved from Companion on Radios without virtual inputs
commit 1251ceea (2016-05-22): Fixes #3510: Fix heli inversion not correctly shown/saved from Companion on Radios without virtual inputs

Also remove comment that might be a todo for exactly this.
PR #3517 (2016-05-31): Fixes #3510: Fix heli inversion not correctly shown/saved from Compan…
…ion on Radios without virtual inputs

*edit (added by projectkk2glider):* Closes #3510
commit 3a63c9ed (2016-05-31): Fixes #3510: Fix heli inversion not correctly shown/saved from Companion on Radios without virtual inputs (#3517)
closed 2016-05-31 by projectkk2glider