opentx/opentx #3513: [lua] logging 'virtual' telemetry values computed by a LUA telemetry script

raphaelcoeffic, 2016-05-19:
While working on my Turbine-ECU-to-SPORT-thingy, I thought it would quite nice to be able to log telemetry values computed by LUA.

Let me extend a bit: these ECUs, just like FCs on multicopter do have status messages which are transmitted as integers (which is what gets logged). Then, in a LUA script, these integers get replaced by a meaningful text displayed on the custom telemetry screen.

Wouldn't it be better to log exactly these messages instead of integer which does not mean anything to the user? I admit these values could not be displayed with the current telemetry log viewer, but we could probably come with something useful here as well.

Later on, with the new LUA telemetry features (sending arbitrary telemetry values directly to LUA), there is just no way to log these things at the moment. Of course, one could probably open a file of its own and basically replicate the logging feature from LUA, but that would be pity, wouldn't it?

Not just limited to text, I guess there would be quite some usage for these 'virtual' sensors. Now, well, I have no clue how to tackle that best, but would implement it if you guys give me some hints.

Maybe it could work just like the mixer scripts with the output array? The logging would then just pick the last value written into the virtual sensor every time it needs to log. We might even be able to overwrite real sensor values for the purpose of computing complex derived values in case the normal functions are not enough.

kilrah, 2016-05-19:
Lua scripts can write files to the SD card, so you'd probably want to do your own logging instead of using the radio's one.

raphaelcoeffic, 2016-05-19:
@kilrah: I'm fully aware of that (and mentioned it in the original post). The question is really whether or not it would useful to have everything at one place, which makes it easier to correlate & display things.

projectkk2glider, 2016-05-20:
This really needs more thought. Here are my random ideas:

The idea of being able to store additional (numerical) values in logs is appealing. 

Any kind of table that will store Lua script values for logging (which might even be text strings) will consume RAM. Not sure users will be happy.

Next problem is how to define which parameter gets into the logging table, its type and how will the Lua script set the value. How will the user be able to select what gets logged and what not?

The suggestion by @kilrah really looks the simplest and most flexible.

raphaelcoeffic, 2017-01-06:
@projectkk2glider just to pickup this old suggestion again: the main issue with the separate logs is that the values need to be correlated afterwards. Even the two logs files need to be correlated afterwards, which could be quite cumbersome for script users.

However, isn't the new setTelemetryValue() function something like that? Creating a new LUA sensor? I guess the values can be logged as well, right?

kilrah, 2017-01-06:
Yes totally, setTelemetryValue() should care for this scenario.

closed 2017-01-06 by kilrah