opentx/opentx #3514: [Bug][OpenTX 2.1.8] approximately equal (a~x) doesn't beahve correctly

LapinFou, 2016-05-19:
Hi Folks,

I went on the field last WE, I had a big issue with my alarms. They don't triggered correctly. Looks like there is a 20% offset.
It is definitely not working as expected.
Here is a testcase:
[image]
[image]
[image]

Here is an EEPE file to reproduce the issue:
[Fuel_Issue.zip](https://github.com/opentx/opentx/files/272623/Fuel_Issue.zip)

raphaelcoeffic, 2016-05-19:
From what I can see in the code, the "almost equal" function does not seem to be designed to compare this type of values.

````
#if defined (PCB9XR) || defined (PCB9XR128)
  #define STICK_TOLERANCE 16
#else
  #define STICK_TOLERANCE 64
#endif

result = (abs(x-y) < (1024 / STICK_TOLERANCE))
````

To me, that means that this function is aimed at detecting stick positions, with a tolerance of 16 to 64 points, depending on the transmitter. In your case, 50 - 35 = 15, which is definitely lower than 16.

LapinFou, 2016-05-19:
OK. Thanks for the explanation, but I think it was working fine before. However I'm not sure. I've not been flying for several months...

raphaelcoeffic, 2016-05-20:
Hmmm.... that's quite strange, as this code does not seem to have changed for quite a long time (at least two years).

LapinFou, 2016-05-20:
I had a look in my backup. Indeed I changed my programming during the winter. So I never tested it. :smile: 
Now, as an end-user, it looks like a bug. What do you think?

raphaelcoeffic, 2016-05-20:
The issue is, I think, that it would be quite cumbersome to implement such a feature for any value range, as the condition function would need to know the value range. I'd prefer restricting/documenting so that it is only used with stick(-isch) values.

The real question here is: what where you trying to do? Different warnings for <35% and <30%? something like "Fuel low" and "Really time to land!"?

hrenz, 2016-05-20:
instead of exact a=x  you have  a~x   "about", "nearly" function
normal its for a range of +/- 1024 ok with  a Hysteresis of +/-3 to 5% 
??????

LapinFou, 2016-05-20:
This is the point of this ticket:  a~x is +/-30% of the Fuel value...

projectkk2glider, 2016-05-20:
@raphaelcoeffic explained pretty well, the hysteresis if fixed and works for values in a range of +/- 1024. Your range is only 0-100.

There is however a trick that might work for you: set the precision of fuel sensor to 0.1, this will narrow the hysteresis to the +/- 1%.

LapinFou, 2016-05-21:
@projectkk2glider 
Indeed, changing the precision does work. Thanks.
I got a question about the precision parameter: how does-it work? So far I leaved the default setting when my Taranis discovered the sensors. I thought changing the precision was useless (if the sensor doesn't send decimal value) and could create some problem for large number. For ex if I set precision to 2 for RPM, then displayed RPM will clipped due to an overflow somewhere.
In other words, could you clarify how does work the precision parameter?

projectkk2glider, 2016-05-22:
Sensor precision works like this:

* every known sensor has a fixed (pre defined) incoming precision parameter. For new sensors that parameter is 0.--. We will call this one **prec_in**

* each sensor has user settable precision parameter. (**prec_user**)

* when new sensor value is received, the value that is stored is calculated like this:
  * if **prec_in** and **prec_user** are the same, then value is not modifed
  * if **prec_in** > **prec_user** then icoming value is divided (either by 10 or 100) depending on the difference of precisions. This means that stored sensor value actually has less resolution.
  * if **prec_in** < **prec_user** then incoming value is multiplied (either by 10 or 100) 

In your case with Fuel, the **prec_in** is 0.--, so when the **prec_user** is set to 0.0- the incoming value is multiplied by 10. So internally the value is 10 times bigger than sensor sent it. When comparing such value in logical switches, the comparison also uses this 10 times bigger value and with the same absolute hysteresis of `a~x` operator this results in narrower range where the LS is true.

bsongis, 2016-05-22:
Is there anything we can do about this issue, it would be nice to change the percentage near ```=```

LapinFou, 2016-05-22:
@projectkk2glider Thanks a lot for this very clear explanation.
I think, as proposed @raphaelcoeffic, the a~x usage should be restricted to avoid such problem.

raphaelcoeffic, 2016-05-23:
@bsongis  or is there a way to find out the range of the input value used in the condition? (sorry for being lazy and not digging into the code...) If yes, then the tolerance could be adapted to the value range automatically. Else, we'd better forbid applying this function to things that might have a different/unknown range.

projectkk2glider, 2016-05-23:
>Is there anything we can do about this issue, it would be nice to change the percentage near `=`

Well, the `v3` parameter is free, 10 bits, it could be used for defining tolerance (either straight v3 value or some kind of progressive scale).

bsongis, 2016-05-23:
Yes this was the idea, the problem is the LCD limitation ...

mhotar, 2016-05-23:
LCD is little bit full, I would say we can consider to do folded multilines. I would see It as an extra line that unfolds itself only when the cursor is on the line. Perhaps we can go without folding and let these lines visible all the time?
[image]

bsongis, 2016-05-23:
Unless we keep one line per switch and we have a submenu?

mhotar, 2016-05-23:
Not a bad idea :)

closed 2020-04-06 by stale[bot]