opentx/opentx #3518: [Companion 2.1.8] [Bug] Wrong assignment of GVar in mix screen

BenZoFly, 2016-05-21:
Hi,

just bumped into this bug in Companion 2.1.8:

1. Create a mix with a GV as weight (not sure if this is relevant for reproducing the bug), say GV3, and another GV as a "diff" curve, say GV5.
2. Save the model.
3. Reopen the mix, and change "diff" by "expo". GV setting is reset. Tick the box again. GV5 appears as the chosen value. Say "OK".
4. Read the Mix screen: The expo is now controlled by GV1.
5. Reopen the mix and re-choose GV5. Now it is set correctly.

When ticking the "GV" box again, and GV5 appears, opening the drop down box and reselecting GV5 again does not help, GV1 will then appear when you click OK. But selecting another GV, like GV4, then selecting again GV5 will then make GV5 the allocated value. I believe this is just a missing variable assignment in the code for when you re-check a GV when it was checked before.

Hope this helps...
Cheers,

BenZoFly, 2016-05-21:
Upon first analysis of file mixerdialog.cpp I see the creation of the variable curveGroup like this:

`curveGroup = new CurveGroup(ui->curveTypeCB, ui->curveGVarCB, ui->curveValueCB, ui->curveValueSB, md->curve, model, firmware->getCapability(HasMixerExpo) ? 0 : HIDE_EXPO);`

It seems that when I change "diff" to "expo" I change the value of curveTypeCB, which resets curveGVarCB to "unchecked" state, but it does not reset curveValueCB, so that when I check the GV box again the previously selected GV is still shown in the dropbox. That could be acceptable, however it seems that the underlying variable (is it stored in the "md" variable?) is reset to GV1.

I cannot investigate further because I am unable to find the ui_* files. IF anyone can point me out to these files, I can continue investigating the bug.

Cheers,
B.

projectkk2glider, 2016-05-22:
The `ui` files have the same name as the corresponding dialog, just the extension is `.ui` instead of `.cpp`. They are in the same directory as the source files that use them.

BenZoFly, 2016-05-22:
Yes I saw the .ui files, but the file I'd like to investigate seems to be called `ui_mixerdialog.h` as quoted in the cpp file:

```
#include "mixerdialog.h"
#include "ui_mixerdialog.h"
#include "eeprominterface.h"
#include "helpers.h"
```

Or if one knows directly where the class CurveGroup is defines, that would be even better!

kilrah, 2016-05-23:
A quick "find in files" (maybe the most important tool for a developer!) shows it's in `helpers.h`.

The `ui_xxx.h` files are autogenerated by QT from the `*.ui`files during the build process, there's nothing to see in them.

BenZoFly, 2016-05-24:
Thank you, hey, first time in opentx code and first time in github :)
I found the issue in `helpers.cpp` indeed:  

At line 278, when you tick the GV checkbox, it sets the curve value to GV1 regardless of its previous state like this:
```
  if (state) {
      curve.value = 10000+1; // TODO constant in EEpromInterface ...

```
Then it calls `update()`, but because it is a change from "diff" curve to "expo" curve, the variable lastType is set to 0 (diff curve) and therefore the re-population of the curveValueCB is not triggered by function `populateGVCB()`. So the curve value was changed, but no one says it to the UI in order to update the index of the combobox.

So as a quick fix, I can propose to insert a new line like this:
```
  if (state) {
      curve.value = 10000+1; // TODO constant in EEpromInterface ...
      lastType=-1; // quickfix issue #3518: force refresh of curveValueCB at next update()
```
With this, the next update will call function `populateGVCB()` with curve.value set to 10001, so it will recreate all values and select GV1. Not very elegant, but it should do the trick for the time being.

When the GV box is not unchecked and the spinbox value is set to 0, the function `update()` resets well the UI, so no issue there.

Cheers!
B.

PR #3532 (2016-05-25): Fix GVar display combobox
quickfix for issue #3518: GVar combobox not being refreshed when GV checkbox (re-)selected

*added by projectkk2glider:* Closes #3518
commit f27162e1 (2016-05-25): Fixes #3518: GVar combobox not being refreshed when GV checkbox (re-)selected
commit 0eda9713 (2016-05-25): Fixes #3518: GVar combobox not being refreshed when GV checkbox (re-)selected (ported from master)
closed 2016-05-25 by projectkk2glider