opentx/opentx #3522: Low Memory Lua Script

ilihack, 2016-05-22:
Hello Opentx comunity,
I Devolop one script  Luapilot for Arducopter,Arduplane,PX4,Cleanflight and Taulabs, Naza

In Simulator everthing is fine but when i load it to the Radio its say low Memory...

the variables are not the Problem...

I hope you can change paramater in Opentx and i can update the new version from the script to githube

the prolematic script:
https://drive.google.com/open?id=0B_jVZeh7dgnIMEtObFZRaWdjTnc

kilrah, 2016-05-22:
Resources on the simulator are infinite, while they are pretty limited on the radio.
Please see other people's takes on the issue in places like here: http://www.rcgroups.com/forums/showthread.php?t=2180477

You might need to separate your script into different ones for each FC type instead of having one big script that does everything.

closed 2016-05-22 by kilrah

lshems, 2016-05-22:
Ilihack, if you PM me on rcgroups, I can explain how to tackle this problem.

Lshems.

bsongis, 2016-05-22:
Or here:
https://gitter.im/opentx/opentx

ilihack, 2016-05-23:
okay thanks you all togher,
@kilrah i comment all other Fc allrady but it low memory still.
@Ishems thank you i try to Write you an pm rcgroup but its not possible you inbox is to full :)

lshems, 2016-05-23:
fixed. I sent you a short mail with contct details