opentx/opentx #3527: Not able to get FW with companion 2.0.19

blairfancy, 2016-05-23:
Following link return error:2
http://downloads-20.open-tx.org/firmware/getfw.php?fw=opentx-9xrpro-heli-fr.bin

projectkk2glider, 2016-05-23:
This is the problem:
```
Fetching Mike's bootloader...
wget http://ersky9x.googlecode.com/svn/trunk/src/bootloader/bootflash8.lbm
--2016-05-23 21:39:10--  http://ersky9x.googlecode.com/svn/trunk/src/bootloader/bootflash8.lbm
Resolving ersky9x.googlecode.com (ersky9x.googlecode.com)... 64.233.167.82, 2a00:1450:400c:c0a::52
Connecting to ersky9x.googlecode.com (ersky9x.googlecode.com)|64.233.167.82|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-05-23 21:39:10 ERROR 404: Not Found.
```
Googlecode project was archived, we need another source.

projectkk2glider, 2016-05-23:
@blairfancy as a temporary workaround, enable `templates` option - this firmware was already build and you will be able to download it. `opentx-9xrpro-heli-templates-fr.bin`

projectkk2glider, 2016-05-23:
@bsongis @kilrah Could we use the urls from the `master` branch for `2.0`?
```
else ifeq ($(PCB), $(filter $(PCB), 9XRPRO AR9X))
bootloader.lbm:
	@echo "Fetching Mike's bootloader..."
	wget https://github.com/MikeBland/OpenRcBootloader/releases/download/V1.9/bootflash8.lbm
	mv bootflash8.lbm $@
else ifeq ($(PCB), SKY9X)
bootloader.lbm:
	@echo "Fetching Mike's bootloader..."
	wget https://github.com/MikeBland/OpenRcBootloader/releases/download/V1.9/bootflash4.lbm
	mv bootflash4.lbm $@
endif
```

bsongis, 2016-05-23:
Yes I think so

kilrah, 2016-05-24:
Yep that would be the one...

projectkk2glider, 2016-05-25:
@bsongis two options now:
* make a new release of the 2.0 branch (perhaps not a bad idea, there is also fix for #3347 since 2.0.19 release)
* or *bodge* this fix into existing 2.0.19 release dir on Jenkins server

projectkk2glider, 2016-05-25:
One more bug:
```
[ 93%] Generating firmwares/th9x/th9xsimulator.cpp
svn: 'http://th9x.googlecode.com/svn/trunk/src' path not found
make[2]: *** [firmwares/th9x/th9xsimulator.cpp] Error 1
```

commit 9d5a74d2 (2016-05-25): Fixes #3527: location for 9XR-PRO and SKY9X Mike's bootloader has changed (ported from master)
PR #3537 (2016-05-26): Projectkk2glider/branch 2.0 remove obsolete firmwares
Closes #3527
closed 2016-05-26 by projectkk2glider