opentx/opentx #3531: Discussion - DS3231 RTC support (for AR9X)

sreichholf, 2016-05-25:
I took my self some time and added myself support for the DS3231 RTC to OpenTX.

I'd like to get that piece of code into the official repo(i know the developer documentation).
BUT as you (those guys actively working on opentx)  probably already know the whole i2c code in the sky9x codebase is, well, let's say "easily improvable" ;).

Considering that I do not have the hardware to clean that stuff up (only got the ar9x board), and probably not enough motivation either I thought it might be a good questions to ask about how to proceed with this.

The current state of my change probably breaks everything BUT the ar9x, but that's easy to fix and I'll definitely do that. 
I have written some quite generic i2c helpers for TWI0 which would most probably be usable if anyone ever decides to clean up the 9XRPro/Sky9X code.

So i would like to finish and clean "my" stuff up and then add a compile switch to enable the DS3231 code for AR9X, (I've seen that the companion is essentially already prepared for something like that)

Would that be OK? (I'll sure let you have a glance at the changes oncethey're ready for that! ;))
I'll give my best to make it as less of a hack as I possibly can without rewriting the existing stuff.

bsongis, 2016-05-25:
OK I will try to help you, I have a 9XR-PRO and a Sky9x board, but no AR9X
You have our full support if you decide to clean any part of the code in our project, there are too many areas that we cannot cover.
Don't hesitate to open pull requests, you will have comments for sure!

bsongis, 2016-05-25:
https://gitter.im/opentx/OpenTX22 if you need to chat

bsongis, 2016-05-25:
I would like you don't do this mods on 'master', but rather on a branch in 'next', I will add you as a commiter so that you will have Travis working with your branches

sreichholf, 2016-06-13:
done and therefore closed

closed 2016-06-13 by sreichholf