opentx/opentx #3535: Horus GUI Theme design

JimMchugh, 2016-05-26:
Just loaded up OpenTX v2.2.0N317 on my beta Horus.

Fantastic!

One question: What is required to create your own theme design?

I'd like a very plain / stark / monochrome design...

3djc, 2016-05-26:
Please join https://gitter.im/opentx/OpenTX22 . I'm closing this as it is not an issue per say :)

closed 2016-05-26 by 3djc

bsongis, 2016-05-26:
I started a TBS theme here:
https://github.com/opentx/opentx/blob/next/radio/sdcard/horus/THEMES/TBS/main.lua
And yes please use the gitter for that, we will try to help (and add more functions to the Lua interface if needed)

dk7xe, 2017-01-25:
When i click the gitter link i get "404. This is not the chat you're looking for. Sorry. :(" any idea?
Just flashed my Horus with OTX and did the painful manual transfer of my models. ;)
but now i'm stuck at creation of own theme.
BTW: if there is an explanation available in a closed group on gitter. Why can't the solution be posted here as well? If you like i can volunteer but first the gitter access issue needs to be solved ;)

3djc, 2017-01-25:
We are not using gitter anymore, we moved to rocket.chat (https://opentx.rocket.chat/). And no, no instructions exist on theme creation