opentx/opentx #3540: HORUS - S1 and S2 pot position indicator jiggling.

DaddyBigas, 2016-05-27:
HORUS
opentx-horus-2.2.0N317

Sometimes my S1 and S2 pot position indicator on the bottom of the main screen shows the pot position as jiggling.

3djc, 2016-05-27:
Have you calibrated those ?

projectkk2glider, 2016-05-27:
I will repeat my answer on gitter here:

>What you see is the result of noise that causes the S1 (or any other) value to change slightly. When that change happens to be right around the tip-over value the graphics change accordingly. So a very small change in input signal produces a 1 pixel change on the LCD. It is more of a cosmetic issue and could be fixed in the future. Also check if you have ADC filter active on the hardware setup page (at the bottom).

Lets leave this issue open to remind us to do the `cosmetic` change to the graphics.

DaddyBigas, 2016-05-27:
Thank you for the answer.  That actually makes complete sense and also explains why it is intermittent.

Yes, ADC Filter is enabled.

closed 2020-04-06 by stale[bot]