opentx/opentx #3541: HORUS - Throttle warning on startup

DaddyBigas, 2016-05-27:
HORUS
opentx-horus-2.2.0N317

I get a throttle warning on power-up even though the throttle is in the lowest position.  I did try moving the stick but the warning remains regardless.  I have done a calibration (multiple times).  I have to press a button to clear the warning.

I have confirmed in the analogs screen that my stick movement is registering correctly throughout its range.

3djc, 2016-05-27:
Have you confirmed the radio is in the right mode ? Mode 1 : throttle stick right, mode 2 : left

kilrah, 2016-05-27:
Yep confirm the mode setting and mention it. Works fine here.

DaddyBigas, 2016-05-27:
That was it...can't believe I missed that!  I have not flown with the Horus yet, I guess I would have noticed that pretty quickly after binding a model.  LOL.  Closing issue.

closed 2016-05-27 by DaddyBigas