opentx/opentx #3544: HORUS Telemetry values not showing in Widgets

JimMchugh, 2016-05-27:
Even when the SWR, RSSI, A1 & A2 value are all showing and changing in the radio setup, when I add a "Sensor" widget to the model screen, no values ever show up...

Widget -> Setting -> Source = RSSi for example

kilrah, 2016-05-27:
Works fine here, so please detail more.

JimMchugh, 2016-05-27:
OK, so I go into the main model page, bind a receiver (D8R-XP) and then "Detect Sensors"

All four "sensors" are found: SWR, RSSI, A1 & A2

I can see the little * strobing though and reading each sensor. Values are changing etc.

Then I go to "Screen Setup" on the main model home screen. Add a Widget which has "Sensor: 1" written on it. Go into "Widget Settings" and select (let's say) RSSI.

The on-screen widget still only says "Sensor: 1" in small font, and doesn't show the RSSI value...

Am I missing a step somewhere?

Cheers!

3djc, 2016-05-27:
Are you sure there are not several screen, and you did setup the widget on anotyher screen ?

JimMchugh, 2016-05-27:
@3djc: Not on another screen. Only one screen setup. Can see widget showing "Sensor: 1" but no values shown...

kilrah, 2016-05-27:
There is no widget in OpenTX that would say this, so you probably got it from somewhere else and it's broken.

closed 2016-05-27 by kilrah

JimMchugh, 2016-05-27:
@kilrah: I understand the close, but what is the widget that shows "Sensor: 1" then?

Also, is there any way of showing a telemetry value such as A1 or A2 on the model home screen?

Cheers!

kilrah, 2016-05-27:
I have no idea, as said no current OpenTX widget would show that. Could be a lua widget you found somewhere and put on the card?

The widget that by default shows "Rud 0" when selected can display any source.

JimMchugh, 2016-05-27:
@kilrah: Yes, you're right! It's the "BattCheck" widget from the current SD Card image...

Sorry!

3djc, 2016-05-27:
BattCheck requires an FLVSS sensor, 

You can display A1, or any 'simple' telemetry value using the widget that display Rud as default, and configuring it to point to your desired telem sensor

kilrah, 2016-05-27:
The BattCheck script that is now in the repo works fine, but it didn't when the SD card image was last made a couple of weeks ago.

JimMchugh, 2016-05-27:
OK, I will replace the CD Card image script with the new one.

Another question: For example I want to show the RSSI value on a Widget. The RSSI value runs between 0 and 100+ but the range in the Widget settings is -1024 to +1024. It takes ages to scroll the values down to 0 to 100. Is there any quick way of setting the value to 0 without scrolling?

Cheers!

kilrah, 2016-05-27:
Not yet.

Banjo2, 2016-12-07:
Hello.
Have you some progress about widget SWR ? /RAS not function, because still show "0"/
Thanks.

sly1231, 2018-06-23:
How does one select 'RSSI' for the top bar widget?  I get the graphical bar in the upper right corner with receiver ON.

sly1231, 2018-06-24:
Never mind - got figured out.

sly1231, 2018-12-07:
I did it once a few months ago - now I don't remember how I did it. I get several choices under "value" but for some reason 'RSSI" does not show up.   I get the graphical bar in the upper right corner with receiver ON.

kilrah, 2018-12-07:
Discover sensors.

Arteggg, 2019-04-27:
JimMchugh is right, i do have same issue. All sensors discovered, but not shown on top of screen