opentx/opentx #3545: HORUS SD Card not appearing in Windows

JimMchugh, 2016-05-27:
With the radio ON, running latest OpenTX fireware, when I connect to a Windows computer (that works fine with Companion v2.2.0) no SD Card shows up in the drive list.

All drivers loaded. Put the SD Card directly into computer and all is fine.

If I connect USB with radio OFF, I get the STM32 Bootloader connected.

Am I missing something obvious?

3djc, 2016-05-27:
The bootloader does not work yet like a Taranis to allow you to access the sd through usb

JimMchugh, 2016-05-27:
But my understanding is that with Horus switched ON, connecting the USB should show the SD Card as a drive.

The radio knows that it's connected via USB as it displays the USB icon in the top right corner...

kilrah, 2016-05-27:
USB storage isn't implemented yet.

Please use the chat for simple questions instead of opening issues.

closed 2016-05-27 by kilrah

JimMchugh, 2016-05-27:
@kilrah: Sorry! Where is the chat at?

3djc, 2016-05-27:
https://gitter.im/opentx/OpenTX22