opentx/opentx #3549: [Bug] trim position change during switch off

BenZoFly, 2016-05-29:
While powering off a Taranis X9D, if a trim is moved while the power off process is being run, the trim sound is audible as if the trim for the last model could still be changed and recorded.

I consider this as a bug (because this is at best confusing the user) and would propose the following:

1/ get the confirmation that the trim of a model cannot be altered during the shutdown procedure;
2/ when this is confirmed, prevent the trim sound from beeping during the shutdown procedure.

I can assist if I am pointed out to the proper code files, I could not find any good lead in my first quick search for evidence.
Regards,
B.

projectkk2glider, 2016-05-31:
The trim change happens in `checkTrims()` function, which is periodically called from `mixerTask`. We can not just stop mixer  task, because it is also responsible for watchdog reset. 

The longest part of shutdown happens in `opentxClose()` where EEPROM is written. This part is called from the `menusTask()` after the exit from endless loop.

There is a watchodg reset code in `eeprom_rlc.cpp`, but I think it is not used in this case. So this needs to be analyzed more to see if we can simply stop mixer task when going into the shutdown. Otherwise a more compicated method will be needed.

BenZoFly, 2016-06-01:
Hi thanks. First analysis: the power off trigger is captured by several procedures, but the one triggering the wrap up of the config, the bye screen and the LCD off is indeed `menusTask()`. So I was wondering what is the watchdog issue with shutting down the `mixerTask()` process from within the `menusTask()` one when it's time to stop the board. I think I understand that the `mixerTask()` can be placed in standby when the `s_pulses_paused` is set to `true`. I would tend to think that it would make sense that when the shutdown process is started, the transmitter stops sending pulses anyway and do not mix anything anymore, also not the trims. Last thing I don't get is why the watchdog would still be needed during shutdown...

I could think about something like this at the end of `menusTask()`:
```

s_pulse_paused = true; // (or call a procedure that does this like pausePulses() from pulses_arm.h?)

#if defined(PCBTARANIS)
  displaySleepBitmap();
#else 
  lcd_clear();
  displayPopup(STR_SHUTDOWN);
#endif 

  opentxClose();

  BACKLIGHT_OFF(); // instead of shutting down the light at the beginning
  // of the power off procedure, because this is very inconvenient in the dark
  // to see when the shutdown process is finished!! (just my opinion)

  boardoff();

```
Did I miss something here? 
Cheers,
B.

projectkk2glider, 2016-06-02:
The problem can happen if we stop the watchdog reset and the EEPROM save (which can take place in `opentxClose()`) takes more than the watchdog timeout. In such case the radio will be reset before the write is complete - probably leading to the EEPROM corruption.

bsongis, 2016-06-02:
We could call ```watchDogSetTimeout(500/*5s*/)```

projectkk2glider, 2016-06-09:
@BenZoFly the idea from Bertrand (to set larger watchdog timeout) is good, use that in your code example before you stop pulses. Your idea about back-light also looks good. Feel free to proceed with this and open a pull request. Thanks.

BenZoFly, 2016-06-09:
OK, I see the logic, I'll propose a pull request on this basis, but will dig first into the watchdog mechanism to understand what I'm touching :)

My Taranis can take a good 5s to shutdown as my eeprom is near full of complex models. I'll seek for a safe timing.

kilrah, 2016-06-10:
I'd rather see something actively resetting the watchdog in the shutdown procedure as the time needed to save to EEPROM is relatively hard to gauge. I've already heard of people with super huge models saying it took 10 seconds to turn off.

bsongis, 2016-06-14:
my proposal:
```
void opentxClose()
{
  watchdogSetTimeout(1000/*10s*/);
  pausePulses(); // stop the mixer
  ...
```

BenZoFly, 2016-06-18:
Hi all, here is my analysis: I have investigated 3 procedure calls possible:
`wdt_disable(); `to completely disable the watchdog (right?)
`pausePulses();` to pause the mixer loop including the watchdog
`pauseMixerCalculations();` to also pause the mixer via its mutex so excluding the watchdog

I am not fond of putting 10s timeout value and pray for it is long enough...

Am I right to say that `pauseMixerCalculations();` would be the safest way to stop the mixer while allowing the watchdog being reset during the shutdown procedure?

The problem with this procedure is that it only do something for arm boards. Other boards will not see the problem solved.

For reference, I found this code in eeprom_rlc.cpp as was pointed out by projectkk2glider earlier:
```

void eeLoadModel(uint8_t id)
{
  if (id<MAX_MODELS) {

#if defined(CPUARM)
    watchdogSetTimeout(500/*5s*/);
#endif

#if defined(SDCARD)
    closeLogs();
#endif

    if (pulsesStarted()) {
      pausePulses();
    }

    pauseMixerCalculations();

```

projectkk2glider, 2016-06-18:
Non-ARM platforms do not have this problem, since they do not use multiple tasks. I am on this right now, I will post a solution soon.

projectkk2glider, 2016-06-18:
I have tried just calling `pauseMixerCalculations();` which means that `wdt_reset()` would still be called from the mixers thread, but it did not work. Apparently not all flags for wd reset are set. 

So in the end the best solution is the one from Bertrand, I have just extended the wd timeout to 20 seconds.

closed 2016-06-18 by bsongis

BenZoFly, 2016-06-18:
Thank you for the fix and for having added the backlight issue in the fix.

Isn't it worrying that the `pauseMixerCalculations()` function is not producing the expected effect???

projectkk2glider, 2016-06-19:
>Isn't it worrying that the pauseMixerCalculations() function is not producing the expected effect???

No, after the second look calling `pauseMixerCalculations()` actually causes the mixer thread to wait for the mutex to release. Therefore the WDT reset is not called.

BenZoFly, 2016-06-19:
Yes of course my fault I misinterpreted the mutex behaviour...
I'll continue to think about it when I have time.Cheers,B.

 Envoyé le : Dimanche 19 juin 2016 10h14

commit 7c19fba1 (2016-06-18): Fixes #3549: disable trim processing in while shutdown is in progress, do no turn of the back-light before the shutdown
PR #3612 (2016-06-18): Fixes #3549: disable trim processing in while shutdown is in progress…
…, do no turn of the back-light before the shutdown

Closes #3549
commit 7fd412f7 (2016-06-18): Fixes #3549: disable trim processing in while shutdown is in progress, do no turn of the back-light before the shutdown (#3612)
commit f1bdb931 (2016-06-18): Fixes #3549: disable trim processing in while shutdown is in progress, do no turn of the back-light before the shutdown (#3612) (ported from master)