opentx/opentx #3550: Jumpy rx values X4R Taranis in Cleanflight

techcoast, 2016-05-30:
So, I have a Taranis with an X4R receiver. Not my first, I've had another and it worked great. Still have it on my first model on the TX. However, I started a new model, binded another X4R and when I went into clean flight, this is all I could get,

https://www.youtube.com/watch?v=UdkvncyV294

SO, I'm putting this out here for comment. I've tried 2 new Rx's with an additional one from a friend who verified that his was working fine. All gave me the same results. The closest I get to anything stable is when I copy the first model, but I still get glitches on the yaw and throttle. 

It's a new TX (Bought it back in March) and it's never been banged around. I'm taking it in to the store where I bought it for a replacement, but I'd like to hear your comments if this has happened to anyone else?

Thanks.

kilrah, 2016-05-30:
It's never been reported at least, I've seen such things on forums occasionally but it was always wrong connections/configuration/version combinations.

closed 2016-05-30 by kilrah

techcoast, 2016-05-30:
A little more direction would be helpful. Since others have had this issue, perhaps some suggestions of what to do? The binding process is very simple, I follow it step by step, I've connected 2 brand new RX (plus 1 that was a friends to verify this issue) to 3 independant fc's after binding and gotten the exact same results. Yet, when I plug in my other quad from a build I did a couple of months ago, everything is rock solid. Same materials, same versions of software, etc.

When I try to copy that first profile to another slot, I get less glitches, bu it's still there. I've bound my new rx's to the first profile to see if that would help, but it still glitches. BUT, when I use the original, it's fine. Something is very, very strange here.

projectkk2glider, 2016-05-30:
If your radio works with one quad, then the problem is not with the radio. Looks the je second naza board is faulty.

techcoast, 2016-05-30:
3 new Naze boards, 2 brand new receivers, and 1 rx that was known to be good, and the problem isn't the Tx??

projectkk2glider, 2016-05-30:
Your writing is hard to follow, I based my comment on this:

>Yet, when I plug in my other quad from a build I did a couple of months ago, everything is rock solid. 

This **really** is not the right place to solve yor issue, please use forums.

kilrah, 2016-05-30:
If you configured all 3 boards wrong, possibly not...

You didn't give any relevant debugging information i.e. firmware versions of everything involved, how you connected things, how you set up the FC and radio, so there's really nothing we *could* even work with even if we knew how your setup was supposed to work.

Wheat we know is that there are many, many, many people successfully using OpenTX with cleanflight quads, so it simply can't be an OpenTX issue which this place is about.

You might have better luck on Cleanflight-related forum threads.

techcoast, 2016-05-31:
Okay, so I'm eating some crow right now. I took it down to my hobby shop and stumped them. However, we did get the exact same results wtih another Taranis. So, this is a problem between the X4R and the Rev6 Naze32. We used a Futaba tx on that FC and it worked fine, we also tried an F3 board to the X4R and Taranis and it was  solid. SO, something to consider for future unfortunate individuals.