opentx/opentx #3552: OpenTX Companion 2.2.0N317

breakcore01, 2016-05-30:
I have a problem with the programm, when i start the programm come this error " Unable to check for updates"

Why is it?

3djc, 2016-05-30:
Please unselect : Automatic check for companion updates, in settings/settings/application settings

closed 2016-05-30 by 3djc

breakcore01, 2016-05-30:
but then i can not download the new firmware for Horus

3djc, 2016-05-30:
yes yyou can, the firmware is another settings, this one is only for companion

breakcore01, 2016-05-30:
how can i download the new firmware for opentx horus? have you a link?

3djc, 2016-05-30:
you need at the same place:
- use OpenTX nightly builds
- Automatic check for OpenTx firmware updates

breakcore01, 2016-05-30:
yes i have make it, but how can i write the firmware in horus?
i have start the horus and then conect to the usb and then when i write the new firmware in horus comes this error , see the photo
[image]

3djc, 2016-05-30:
The drivers have for horus have not been installed on your pc