opentx/opentx #3560: Select MIN, MID and MAX for OverrideCH

fraternl, 2016-06-01:
I believe that OverrideCH receives its criticism mainly because it's able to override the endpoints set in Outputs. 
If the identifiers MIN, MAX and MID can be chosen as parameter (after long-press) then that would be solved.

It would then still be possible to override it with numerics.
Clipping to the endpoints could be an idea, but I don't think that's a good one.

LapinFou, 2016-06-08:
You forgot to mention that the OverrideCH doesn't take either in account the direction programmed in "Outputs".

schwabe, 2016-06-08:
Probably if the OverrideCH would be a bit more aptly named that could avert confusion. E.g. overrideOutput. I don't think that MID,MAX, MIN are good choices as you still have the ambuguity if that values are the normal/extended position or the positions that are set in the output dialogs.

@LapinFou Yes, but it is designed to set exactly a value.

projectkk2glider, 2016-06-09:
I would personally just remove the `OverrideCH` function completely. It is already optional (one of the firmware options) and only keep in the source in order to keep existing users happy. Related issues #3398, #1428

LapinFou, 2016-06-10:
@projectkk2glider What he said! :smiling_imp: It is a dangerous option for people who are not OpenTX expert. I would personally not remove the option, but simply enable the **nooverridech** option by default.

kilrah, 2016-06-10:
I would definitely not, because the vast majority of files that are being shared online rely on it and would have unwanted behavior. It's already a problem for those who tick the nooverridech option, and I've already read about incidents due to that.

3djc, 2016-06-14:
I had a rather lengthy discussion with glider competitor (F5a F5j) and they have the following need:
They are getting their glider ready, and get inline and wait for an official to have launch go. When they have that go, they need to be quick (you do not have time to connect the lipo, check, close canopy,..).

So they have special ESC, that work as standard ESC, and brake at zero throttle (for prop folding and drag). The issue is that the brake on feature consumes energy, and their lipo are sized for a single climb. So those special ESC release brake when they get signals well below calibration values, so they use override/cut to set it to -150 and reach that ultra low consumption zone

closed 2016-11-27 by fraternl