opentx/opentx #3569: lcd.drawNumber invers formatting does not invers the dot

richarddeweerd, 2016-06-05:
When you use the invers formatting tag on lcd.DrawNumber the decimal dot is not inverted.

The red circled part in the attachment is a result of this part in my LUA script, where the current value of "Voltage" is 0.2 volt

`lcd.drawNumber(154, 2, Voltage * 10,PREC1 + SMLSIZE + INVERS )`

[image]

3djc, 2016-06-06:
I was not able to reproduce: 

`local function run(event)
	lcd.clear()
	local myVoltage = 0.2
	lcd.drawNumber(154, 2, myVoltage * 10, PREC1 + SMLSIZE + INVERS )
end
return { run=run }`

runs perfectly on my Taranis, 2.1.8

richarddeweerd, 2016-06-06:
Hmm maybe because I draw the black bar first and then overwrite it with the drawnumber. I will check that later today when I'm at home, I can't debug it now. 

BTW. I also run it on my Taranis with 2.1.8

3djc, 2016-06-06:
Indeed, this code exhibit the issue too:
```
local function run(event)
	lcd.clear()
	local myVoltage = 0.2
	lcd.drawFilledRectangle(1,1,211,10, SOLID)
	lcd.drawNumber(154, 2, myVoltage * 10, PREC1 + SMLSIZE + INVERS )
end
return { run=run }
```

closed 2016-06-08 by bsongis

projectkk2glider, 2016-06-09:
This commit 177f672619c0b6655c08f97b6b8ce6e2d2f58ee8 is the fix for the `next`, the commit comment is unfortunately wrong.

PR #3577 (2016-06-08): 3djc/#3569 master
This fixes #3569 on master (next already fixed and merged)