opentx/opentx #3574: FrSky - A3 (and A4?) scaling not correct in 2.1.7/8

MikePRoberts, 2016-06-07:
FrSky changed the range for A3 and A4 data to be 0-330 rather than 0-255 as for A1 and A2.  The net effect is that when (for example) one enters 25.5 for the ratio value then the readings go from 0 to 33.0v.  This did correctly go from 0 to 25.5v on 2.0.x.

Words fail me as to why FrSky made this inconsistency that only adds confusion (actually they don't but they are not printable).

Many thanks for your outstanding work.  Let me know if I can help by checking the revised code.

Mike

MikePRoberts, 2016-06-10:
Is there any way that FrSky would be persuaded that scaling analogue data at the receiver makes no sense as it will have to be scaled again at the transmitter (dependent on the resistor ladder used)?  The raw value will have a range of 2^n-1 and it seems most logical to send it unadulterated.  Mike

kilrah, 2016-06-10:
The problem is precisely that "n" is different in both cases...

This is due to D series system transmitting the analog values as a single byte, i.e. full scale is 255.
X series have 32bit values for everything, so to increase resolution and match other value types they send volts directly. 

We have to process both differently, no choice. Now the choice is what do we base ourselves on? I believe in 2.0 we used A1/2 as reference and scaled back A3/A4 as soon as received and before any further processing to match it because everything else in telemetry code was expecting a single byte, but on 2.1 we use the new, better system as base and scale A1/A2 to it. Or if we don't we should.

fpvflyby, 2016-06-15:
I have a naze32 revision 6 hooked up with S Bus and I am running telemetry on soft serial pinS 5 & 6 it is only sending one cell volts to my Vfas on my taranis I'm using the x4r sb.

kilrah, 2016-06-16:
That's got nothing to do with this issue, nor with OpenTX for that matter.
See cleanflight docs and forums for help about that.

closed 2020-04-06 by stale[bot]