opentx/opentx #3579: [Enhancement][OpenTX 2.1.x] New calculated sensor: electric fuel gauge

LapinFou, 2016-06-08:
Hi folks,

I'm using the UniSens-E sensor. It is a great very compact sensor. One feature which I really like is the "fuel" feature. When programming the UniSens-E with its USB dongle, you can setup a LiPo capacity. Then the UniSens-E will transmit a fuel information from 100% (LiPo full) to 0% (LiPo empty).
This is very, very convenient to setup some warning/alarm (in my case a warning at 35% and an alert at 30%). The major inconvenient is: you need to used the USB dongle to change the LiPo capacity.

I realized that almost everything is present on OTX 2.1 to achieve the same feature.
There already is a "consumption sensor" (Calculated → Consumption → name of current sensor).

Would it be possible to add an "***electric fuel/gauge***" calculated sensor?
**Calculated → Electric fuel (or gauge) → name of current sensor → max capacity of the LiPo**
The unit will be %
The sensor will start at 100% (consumption=0) and decrease to 0% (consumption=capacity).
The basic computation will be something like:
_1 - Consumption / Capacity_

The nice thing to be able to do it from OTX, is be able to create several calculated sensors with several different LiPo capacity.
Then, by using LS with the additional AND, we can select the appropriated capacity, but the alarm logic (and associated SF) will always be the same. It will even be better!! :smile: 

What do you think?

schwabe, 2016-06-08:
I think it probably better to have a scale/divide option in calculated that lets user multiply a sensor by a fixed number. E.g. Multiply, Source1 battery consumption, source2 0.04545 (=100/2000) unit %. Or add a new sensor type that is "Fixed" and always reports the same value and then use that as source2 in the calculated sensor.

Basically the same as you electric fuel but more aptly named and more suited for other functions.

LapinFou, 2016-06-08:
@schwabe With your proposal, you will generate a percentage from 0% to 100%. It is not a gauge. It is a conversion from mAh to % (off course depending of the LiPo capacity).
What I'm requesting is a gauge going from 100% to 0%. Same thing as in a car.
Personally, I'm display this information using the "bar" screen type. It is very handy. You don't need to actually read the value. A quick look on the screen gives you the information you need.

3djc, 2016-06-08:
I think it is way to prone for errors, alarms on volts are way better, because they don"t care about lipo size (which btw, varies greatly with age/condition)

LapinFou, 2016-06-08:
@3djc  No! You're not right. The best indicator is definitely the consumption. This is very accurate source for alarms. The voltage does vary a lot under load and temperature. It is not a good indicator.
About age/condition, you're right there is a variation because the internal resistance of the LiPo will increase. Thus, a part of the energy will be transform in heat instead usable current. But this is also true for voltage monitoring. With the increase of the internal resistance, the voltage drop will also be higher.

3djc, 2016-06-08:
Let's agree to disagree : you cannot be 100% sure you did put a fully charged battery in (voltage will tell you). If only one cell goes berserk consumption will tell you you've got plenty of juice (when voltage will tell you time to land !). After playing over a year with vfas sensors, from my experience, it is not a safe way to ensure you can fly safely, when FLVSS did never ever let me down

LapinFou, 2016-06-08:
> you cannot be 100% sure you did put a fully charged battery in (voltage will tell you)

True
> If only one cell goes berserk consumption will tell you you've got plenty of juice

True
> After playing over a year with vfas sensors, from my experience, it is not a safe way to ensure you can fly safely, when FLVSS did never ever let me down

OK. Here your are talking about safety. In this ticket, I'm talking about "energy tank".

If we are talking about: "**what is the best sensor to detect when a LiPo is empty?**" The answer is VFAS sensor. Detecting "**how much energy is remaining in the LiPo**" is definitely a job for the VFAS, not by monitoring the voltage.

If we are talking about: "**what is the best sensor to detect the health of a LiPo?**" The answer is FLVSS sensor.

So, they are complementary.

FLVSS does more than VFAS, but if you want an **accurate** information of the remaining energy, VFAS is the solution.

projectkk2glider, 2016-06-09:
@LapinFou adding new type of sensor would require an EEPROM change. 

I would just write a simple Lua mix script that would calculate battery lever from 0% to 100% (or whatever you want) from the current value of consumption sensor and some fixed mAh value. Or you can be even more creative and set target battery mAh value from Lua telemetry script or file....

LapinFou, 2016-06-10:
@projectkk2glider OK, I didn't know this request will imply such major modification. :astonished: 
Yes, LUA is the solution. I was hoping to have it added directly in OTX... 
If this feature is only for a couple of persons, then the team shall not spend some time on it. U thought it could be a cool feature. :smirk:

kilrah, 2016-06-10:
I'm pretty sure you can already achieve that with the current system and clever use of calculated sensors...

LapinFou, 2016-06-10:
I don't see how, since only sensors can be chosen in the fields. To achieve this, you need at some point to enter a fixed value somewhere.

kilrah, 2016-06-10:
I'd create a new "custom" sensor with a non-existing ID and enter the value as offset.

Although that could fail due to the sensor becoming "old" as it's never received. In this case we should really consider a new Calculated sensor type for constants.

LapinFou, 2016-06-10:
Offset is limited to 300.

schwabe, 2016-06-10:
I will make a pull request/suggestion for a constant _calculated_ sensor.

kilrah, 2016-06-10:
The 300 limit is a bug in companion, the radio allows more. Already mentioned in another issue

LapinFou, 2016-06-10:
Sounds good! :smile:

LapinFou, 2016-06-10:
I've been able to achieve this feature with a "mixes" LUA script. But, since it is not a calculated sensor, 2 things are missing:
1. I cannot specify the unit (%)
2. It is not possible to log it on the SD card

projectkk2glider, 2017-01-14:
Lua can now set values of custom sensors, which is a workaround for logging . The constant calculated sensor is also implemented. Closing.

closed 2017-01-14 by projectkk2glider