opentx/opentx #3581: Logical Switches: a>x etc.: x limited to le. 150 if a is the output of a lua script

pau33, 2016-06-08:
For me, this is an interesting behaviour. I tried to set up a logical switch, which is triggered, if the calculated distance from a LUA-Script output is larger then 250(m). But 150 is the biggest number I can fill in for x. With other parameters for a (for ex. SWR and others) higher values for x are possible.
Would it be possible to increase the x-values for lua-scripts to higher values as well?

Best Regards Paul

projectkk2glider, 2016-06-09:
Similar problem on the radio, where the biggest value is 100. The limit could be much bigger.

kilrah, 2016-06-10:
Just noticed something related now, in companion offset for telemtry sensor is limited to 300 for no good reason, radio allows much higher values.

pointu, 2016-06-12:
We have the same problem with the GV, limited to 1024, so when u use for exemple Alt (vario altitude, it's à prec1 number) and do something like alt limitation (if Alt >GVx do something) u are limited to 102,4 m.
A work around is to do a calculated sensor to retrieve it in prec0 (m) and the limit is 1024.

PR #5445 (2017-12-19): [Radio] Adjust value limits for mix sources and update UI for LSs, CFs & telem.
Adjust some value limits for mix sources and update editing UI for LSs, CFs, & telem. bars:

  * Increase timer limits to +/-8:59:59 when used with LSs, CFs, & bars;
  * Increase Lua script output limits to +/-30K (see #3581);
  * Trim limits based on extended limit model setting;
  * Limit TX voltage to 25.5;
  * Limit TX time comparison value to 29:59m;
  * Use proper GVar limits (user-configured), display precision, & unit in LSs & telem. bars;
  * Display timer values properly with hours (instead of mmm:ss);
   [... cut]
closed 2017-12-19 by bsongis