opentx/opentx #3584: Companion model window does not format in same way as other windows.

Oldgibber, 2016-06-10:
A fairly trivial point, but should the model window on the Companion take the user's windows formatting?

[image]

The issue is fairly obvious when the window colours are changed to red. This is Windows 10. 

All other windows seem to follow user formatting.

projectkk2glider, 2016-06-13:
@Oldgibber the model list window is different type from the rest of the windows. What color is the main Companion window in this case: red or blue?

Oldgibber, 2016-06-13:
It is red as in the General Edit window above. I thought it was a different type of window, its not scalable either like the rest.

kilrah, 2016-06-13:
The model list window indeed sticks with a "Windows 7 like" style whatever the system is set to.

elecpower, 2016-07-15:
It seems the model list window behaves different to more prominently highlight the 'active' models list when multiple model files are open. Seems to be using one of the current os theme colours. In some themes the difference between active and inactive widows is very subtle and maybe the decision was to make it more obvious for the models list windows.

mpaperno, 2017-02-20:
Apparently this is a "Windows thing."  MDI areas are just not "cool" anymore I guess so MS never updated the theme support (even in Win7 they never truly followed the Aero themes but it didn't become obvious until Win8, and now glaring with the Win10 themes).  This is not specific to Companion or even Qt.

There's a lot of info on the 'nets about the issue, but this post summarizes it nicely.
https://forum.qt.io/topic/63224/qmdisubwindow-in-windows-10

So either stop using MDI, add/use custom styles for the whole app, or just live with it.  Personally I think the MDI paradigm works well in this case.

Closing this unless someone has a better idea.

closed 2017-02-20 by mpaperno