opentx/opentx #3591: interface issues using usbasp (v2.0)

Goofieshot, 2016-06-12:
in the newer versions of companion (2.018,2.019) i have not been able to read my tx. I still have the 2.017 version installed and has no issues reading and writing to the radio. Is this a usbasp issue?, should i be going to a different interface if i wish to upgrade? - Thanks in advance for your help
[image]

lepicfoto, 2016-06-12:
i had somthing off a simelar problem, that is until i realize that i had installed  ver 2.#1#.8 and not 2.#0#.8, iI upgradet all to 2.1.9 both comp and trans via companion, and then all is ok,
Just a hint, it looks a lot like my prop, version mis match.
just a thought

kilrah, 2016-06-12:
Indeed make very sure of tzhe versions of both companion and firmware, and also... mention what radio you're using.

Goofieshot, 2016-06-12:
radio used is 9xr, and you are correct - I have discovered that new firmware upgrades do cause a conflict if the companion version is not upgraded. I didn't clue in to that since the companion ver. allows you to upgrade to the next firmware ver. without upgrading the app. Thanks for your help.

closed 2016-06-12 by Goofieshot