opentx/opentx #3594: GPS is unable to handle negative heights

fraternl, 2016-06-13:
Half of our country is beneath sea level and when I use my GPS there (uBlox, SP Racing F3, S-port to Frsky XSR) the height is clipped at 0, but also shoots at 65535 sometimes.

I thought it was cleanflight's fault, but they write it's properly handled by them.

Could you examine where it's done wrong?
I can only test a few days from now as I'm going on holiday next weekend.
https://github.com/cleanflight/cleanflight/issues/2223

It should only be a use of "int" where now an "unsigned int" is used.

https://github.com/cleanflight/cleanflight/issues/2223

3djc, 2016-06-13:
https://github.com/cleanflight/cleanflight/blob/e133331cf34f94109774181c2d6f3ae74f4fba05/src/main/io/gps.h

Line 110:
extern uint16_t GPS_altitude;              // altitude in 0.1m

Looks like they are handling it unsigned; I believe OpenTx handles negative GAlt

3djc, 2016-06-13:
What is displayed in cleanflight configurator when you have a GPS lock ?

fraternl, 2016-06-13:
I've never seen that...
Where I have my computer I can't get a GPS-lock and if I could it's on the first floor.

I will test something this afternoon using a laptop.

fraternl, 2016-06-13:
I just now tested it in Cleanflight configurator and there it is reported as 65533 metres.....
I think I will head over to the Cleanflight section....

projectkk2glider, 2016-06-13:
Closing then, reopen if the problem is indeed in OpenTX.

closed 2016-06-13 by projectkk2glider

fraternl, 2016-06-13:
I can't judge it properly.
It seems they insist it's in OpenTX

bsongis, 2016-06-13:
Do you need a firmware with LOG_TELEMETRY=YES ? 
There will be no doubt ...

fraternl, 2016-06-13:
Hi Bertrand...
I assume that will log the raw data.
Yes, please....
I'm currently running today's nightly

kilrah, 2016-06-13:
Could we not have an issue here? 
https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky_sport.cpp#L188

Seems everything we receive is considered unsigned?

fraternl, 2016-06-13:
And how does latitude/longitude work?
These can be negative as well or is that carried in another field?

schwabe, 2016-06-13:
I think that signed is converted later. @fraternl does your flight board have a barometer? 

My cleanflight board has one and the (relative) altitude is sometimes negative.  I remember that there was some version (2.1.6 or 2.1.7) that a a broken altitude that showed -4,2 million.

And I cannot find a difference in GPS altitude and normal altitude handling in OpenTX at the moment.

fraternl, 2016-06-13:
Yes, I have a barometer and it can go negative...

fraternl, 2016-06-13:
I can wait some more for that "LOG_TELEMETRY=yes" version.
But if that takes a bit longer I'm going home to eat and I can test it tomorrow.
I'm sorry to admit that I have no development environment for OpenTX.

schwabe, 2016-06-13:
@fraternl I looked into the cleanflight source and I think I found the problem. I commented on the Cleanflight bug tracker

projectkk2glider, 2016-06-13:
@schwabe you are correct :+1: