opentx/opentx #3595: Persistent not always persistent

fraternl, 2016-06-13:
I'm using the persistent flag for consumption in the telemetry settings.
This works great, but sometimes, after charging the battery I want to check how the consumption relates to the amount I just charged.
When I turn on the radio this value is not displayed.

It will only display if I turn my receiver on.

I need to be quick then, because I have an SF that resets the consumption if it detects a full battery.

It would be nice if persistent values (or maybe all values?) continue to be displayed.

projectkk2glider, 2016-06-13:
There is a workaround (I believe):  Telemetry model setup page will show persistent value even if telemetry is not being received.

fraternl, 2016-06-13:
No, 
I was already looking there in the first place.

bsongis, 2016-06-13:
@projectkk2glider do you take it?

projectkk2glider, 2016-06-13:
ok

closed 2016-06-13 by projectkk2glider

fraternl, 2016-06-15:
Works!!!

commit d5e4f113 (2016-06-13): Fixes #3595: show value of persistent sensor after power on, before the telemetry is received
commit 3eda58c9 (2016-06-13): Fixes #3595: show value of persistent sensor after power on, before the telemetry is received (ported from next)