opentx/opentx #3596: Saving and restoring plane builds - companion 2.019

Goofieshot, 2016-06-14:
hi again - I have clean installed the latest companion 2.019 and the latest firmware for it. The issue that i am running into it is, that every time i start a new model build and save it, i cannot re-open it. 
[image]

This I noticed is similar issue as described in #3333 post, but I did not grasp the resolution. I am running 9xr radio, and as per opentx opening write-up branch 2.1 is aimed at the taranis xe radio and the 2.0 is for the rest.
The companion did read and write the latest firmware to my radio successfully, it seems to save the file to a location, it just will not reload the saved file back for further editing. 

 Any suggestions? thanks in advance.

kilrah, 2016-06-14:
> as per opentx opening write-up branch 2.1 is aimed at the taranis xe radio and the 2.0 is for the rest.

That is incorrect. What the notes say is that the X9E cannot run anything before 2.1, but 2.1 can run on all platforms. The solution for #3333 apparently was precisely to move to 2.1 which should have the issue fixed.

closed 2016-06-14 by kilrah

Goofieshot, 2016-06-14:
so what you are saying is that I am good to go to 2.1.8 with 9xr radio