opentx/opentx #3598: 9xr radio thr and ail switch always armed

Goofieshot, 2016-06-14:
Ver 2.1.8 - Finally got this ver somewhat figured out except for these two switches. Had this issue in  a earlier ver and all i did is deselect the frsky build option to get back to business. This time I had no success even with frsky deselected. Do i have any other options? thanks for your help.

Goofieshot, 2016-06-14:
when are the radio settings and build options written to the radio? with firmware write or model write?

mhotar, 2016-06-14:
You should select your build options in companion, then re-download firmware file, and re-flash it.

Goofieshot, 2016-06-14:
didn't work but thanks - set up the radio in options and re-downed firmware, installed it and still am in telemetry mode with switches set. This build option is getting loaded because in the radio all models show telemetry setting page (page 10).

Goofieshot, 2016-06-14:
btw i did reload all models after the new firmware load

Goofieshot, 2016-06-15:
i generated default firmware with no build and default splash and for some reason the radio did not take it because upon completion i had the same old splash screen. Companion did say that it completed it successfully. The only error i get in the beginning of write points to the usbasp

[image]

but it seems to write models and models and model changes in fine - now I'm really confused

Goofieshot, 2016-06-15:
in the end i loaded the same ver of companion on another win7 machine and it did everything fine - imagine that. Wrote the firmware and models without any issue - i apologize for the confusion, and would like to thank for the support. In conclusion at least I have acquired some knowledge on the companion soft.

closed 2016-06-15 by mhotar