opentx/opentx #3601: vfas incorrect

fpvflyby, 2016-06-15:
im running naze32 rev6 with x4r sb on sbus. im using softserial 1 for telemetry. my taranis is only reading one cell of my pack. all other telemetry is fine.do i need to update to open tx 2.1.8 for the telemetry to work right or could i have another issue. i have an x4r sb on my spracing f3 board and all is good.

schwabe, 2016-06-15:
That sounds like naze32 is only reporting the value of one cell.

closed 2016-06-15 by schwabe

fpvflyby, 2016-06-15:
When I connect to clean flight it shows the correct voltage. If I reinstall the flight controller firmware do you think it could fix the problem.Or does the naze 32 revision 6 have issues

schwabe, 2016-06-15:
@fpvflyby that question is rather suited for Cleanflight/baseflight or whatever you are running on the naze32. But all details you are providing are pointing to the naze32 board reporting a wrong vfas. But since it is the voltage of a single cell that sound like it is done on purpose.

kilrah, 2016-06-15:
You did not say what telemetry value you were looking at on the radio. As said, it should be VFAS.

fpvflyby, 2016-06-15:
It is

fpvflyby, 2016-06-15:
Can it be changed in the cli

kilrah, 2016-06-15:
Wasn't a new feature added in CF at some point where you could choose whether the transmitted voltage was "total voltage / number of cells" instead of "total voltage"? You might have enabled that.

schwabe, 2016-06-15:
Iirc my cleaflight board reports both, as a4 and Vfs (or something similar)

fpvflyby, 2016-06-15:
Im running betaflight 2.7.1