opentx/opentx #3603: companion firmware flash usbasp 9xr

Goofieshot, 2016-06-15:
I would like to inquire if this is normal - when i fresh install the ver - 2.1.8, and generate a firmware with tx settings, the companion will let me write it to the radio. If i change my mind and use a new splash screen, regenerate the firmware with that splash screen and try to flash - the tx will be read during the write process but the write will not occur. To be able to make it write again i have to un-install the opentx and reinstall it again. The fresh install will again allow me a one time firmware write. This only affects the firmware write, the model upload and download from the tx behaves normally. Is this a bug or is the firmware check smart enough not to overwrite itself?. Thanks for your info

kilrah, 2016-06-16:
This is an issue with some USBASPs where when 2 operations are done close to each other the 2nd one fails. Reinstalling has nothing to do with this, only that the time it takes you to do so is long enough for the issue to disappear. 
You might want to use another usbasp that works correctly, or alternatively disable the automatic backup and firmware checks in companion as that will thus not do multiple operations anymore. Of course take care of doing backups yourself then.

closed 2016-06-16 by kilrah

Goofieshot, 2016-06-16:
removing and plugging in the usb did not yield the normal response, same with a computer restart. Hence why I had all the issues with getting rid of the flash with telemetry build attached to it. Just for curiosity I left the tx alone while i was at work and upon my return I tried again with no success. That is why i removed the soft and reinstalled it. I did disable the updates also with no luck - but if you think that's a usbasp issue then I will endeavor to look for another. Thank you.