opentx/opentx #3609: Request: add Function Script "background" function

lshems, 2016-06-18:
It would be nice if the Function Script would allow to have a background function as in a telemetry script, if the switch condition is not true.

This would also solve the problem of knowing if the function has been deactivated and activated again by moving the associated switch

I have a custom countdown function, that is activated on Adown. it needs to pause on A-, and needs to reset on Aup. 

the pause is already difficult to achieve, but possible with some time checks to see if the function has not been called and then not update the countdown counter

the reset is now impossible to implement without a background function, since the script is not active as soon as Adown is moved up !

It would also allow telemetry scripts to be run under a function script, and use the switch to toggle functions. (there is no display function available in function scripts, but for voice commands that is not neccesary)

bsongis, 2016-06-18:
Yes agree

projectkk2glider, 2016-06-18:
Isn't the Function script called only once each time when the LS changes from 0 to 1? That I think is the point of Functions script, to be executed exactly once on LS trigger.

You might be better of using a Mixer script, which runs continuously, You can monitor unlimited number of switches is such script and react accordingly. The script runs non-stop.

bsongis, 2016-06-18:
the ```run()``` function is called when the function is active
the ```background()``` *could be* called when the function is in the list, but inactive

projectkk2glider, 2016-06-18:
You are right about the `run()` and you proposal is good.

commit 38fdae6f (2016-06-22): Fixes #3609 - ```background()``` function in Lua function scripts
closed 2016-06-22 by bsongis