opentx/opentx #3615: Website Link Update Request Lua

rguptaca, 2016-06-19:
On this page
http://www.open-tx.org/lua-instructions.html
Could this link be added
https://www.gitbook.com/book/opentx/opentx-lua-reference-guide/details

On this page
http://www.open-tx.org/documents
Could this link be added
http://www.open-tx.org/lua-instructions.html

On this page
http://www.open-tx.org/
Could this link be added
http://www.open-tx.org/lua-instructions.html

Suggestion:
Remove documentation links from everywhere on the site and place all documentation links only on the documentation page. This helps beginners access the most recent documentation. For long time users they can still have access to older documentation if they are running older versions.

I am willing to volunteer my time to help look after the website.

projectkk2glider, 2016-06-27:
The open-tx.org website is generated from https://github.com/opentx/opentx.github.io repository. You could make the changes you suggest and submit a pull request there. In fact this issue should be posted there and closed here.

rguptaca, 2016-08-07:
Opened https://github.com/opentx/opentx.github.io/issues/8

closed 2016-08-07 by rguptaca