opentx/opentx #3616: When using the internal Module on Taranis play 9D w/ crossfire - no telemetry

Webbsolution, 2016-06-20:
I have installed a crossfire module on the internal bay - I want to use BOTh internal module for 2.4 AND external module for UHF -- when I turn the protocol to crossfire I lose telemetry. RCgroups moderator suggested I post this as an issue. If there a way to preserve internal Telemetry on X series RX's while using the internal module with crossfire?

schwabe, 2016-06-20:
Since the taranis shares the telemetry pin (which is used for crossfire) between the internal and the external module this unfortenately a wontfix/cannotfix.

kilrah, 2016-06-20:
As explained on RCG this is out of our control, the only way is for the module manufacturers to do things differently or to hack the hardware.

closed 2016-06-20 by kilrah