opentx/opentx #3623: BUG? using lua to set SpecialFunction to run a script doesn't initialise script on switch true

lshems, 2016-06-21:
If using LUA to assign a script to a special function, this script doesn't get initialised, not even when the SF trigger switch becomes true.

So it doesn't become active until transmitter restarts or other action that retriggers script initialisation.

When assigning it manually from the SF page, it does get initiated.

lshems, 2016-10-27:
Added info: When deleting a trigger from the SF assigned to a lua script, the lua gets loaded nonetheless. This consumes unnecesarry resources.

bsongis, 2016-10-27:
Perhaps you could give an hand and fix this one, it's an easy one, I am
sure you could try to fix it and open a PR

lshems, 2016-10-27:
Look, I really would like to, but I have spend several days trying to get a compile toolchain working, and didn't succeed.

I work on windows 10, and have no other experience than LUA, based on 35 year old memories of BASIC on a commodore.

Sorry, its not so simple for me. 

I think, and you know that, that the LUA interace is the most powerful part of all OpenTX, next to the basic framework of setting up the inputs, mixes and outputs, what I would call operating system.
I would love to add things like LUA functions to change a curve, or whatever.

bsongis, 2016-10-27:
Why don't you install VirtualBox with a Linux VM, it's really simple, you
will find help here for that!

3djc, 2016-10-27:
Once you have a Debian Jessie installed, we can get you to succefull compile in less than 10 mins

lshems, 2016-10-28:
Appreciate the help. A  linux environment setup on VM is not that difficult. I don't like linux at all, but it's just for coding, so i'll give it a try. 

Will be after holidays. family duties :)