opentx/opentx #3627: Cannot bind in + polarity to Spektrum receiver.

zombiediesel, 2016-06-24:


kilrah, 2016-06-24:
Doesn't mean a thing. What module?

zombiediesel, 2016-06-24:
Spektrum dm9 trying to use ar7010 receiver. The thing is, my friend has the
stock taranis firmware on his and it works. I don't want to downgrade
though, because I have frsky telemetry setup on several other models.

<https://github.com/opentx/opentx/issues/3627#issuecomment-228465189>, or mute
the thread

schwabe, 2016-06-24:
What is your version of OpenTX and what version does your friend use? Do you use the same settings for the ppm output as your friend?

jtaylor2, 2016-06-24:
Does it work with - polarity?  If so use minus.  There was a change to the ppm output polarity in version 2.0.16.  Below is from the release notes:
 
Version 2.0.16 / 2015-05-03
[Taranis] 
Fixed reverse polarity on external PPM output. Users with external module in PPM mode should check and update PPM polarity setting.

zombiediesel, 2016-06-24:
I'm using 2.1.8 and he never upgraded firmware from when he bought his a
couple years ago. Ppm settings are identical. His works, mine doesn't.

zombiediesel, 2016-06-24:
The receiver will not work with - polarity. The manual specifically calls
for +

jtaylor2, 2016-06-24:
Try these settings for 2.1.8.  I don't have one, but this is what has been reported to work.  Firmware below 2.0 16 needed + polarity.  And verify that this is an actual DM9 and not a hack module in a DM9 case.

External RF Module in the Model Setup menu:
PPM
CH1-8
22.5ms
400u (not the default 300)
- polarity

zombiediesel, 2016-06-24:
That is what mine has been setup as all along. Does not work. Again, the
only difference between mine and my buddies is the polarity setting.

schwabe, 2016-06-24:
Did you swap the modules to check if both modules show the same behaviour (your works in his, his does not work in yours?)

zombiediesel, 2016-06-24:
Yes I did. The only variable I havent tested was to flash his radio to
current firmware, but he wouldnt let me do that. I have seen this issue
reported about previous versions of the firmware as well.

kilrah, 2016-06-25:
As mentioned your discrepancy is that the polarity display has been changed form one "convention" to the other in 2.0.16, and whatever instructions you're reading might have been written before that or use what was used before.

Just use the polarity that works.

closed 2016-06-25 by kilrah

zombiediesel, 2016-06-25:
The problem is, neither one do...