opentx/opentx #3628: DSM protocol not working on 9Xr pro and nigthly build

blairfancy, 2016-06-25:
Hi,
I m not able to bind my DSM2 heli with 2.1.8N786 on 9xr pro radio
PPM signal is coming out PPM pin.
With 2.0.19 no issue bind every time.

schwabe, 2016-06-25:
What module are you using?

blairfancy, 2016-06-25:
I use module made as follow
http://openrcforums.com/wiki/index.php/9x_Full_Mod_DSM2/X

schwabe, 2016-06-25:
Then you need to select the DSM2 protocol instead of PPM.

blairfancy, 2016-06-26:
Sure I did it, module is working properly on 2.0.19. It is only a regression. I see signal on ppm pin

blairfancy, 2016-06-26:
Signal scoped on pin ppm of module 

[image]

schwabe, 2016-06-26:
Sorry. Your first report sound like you were using ppm instead of dsm2 serial protocol.

projectkk2glider, 2016-06-27:
That signal looks very strange, perhaps it has something to do with the **output type** setting not being visible in 2.1.8 (was fixed in 2479aebde54744463f94642c000d4cd1412b39e9 and will be available in 2.1.9).

projectkk2glider, 2016-06-27:
Scratch that, I see that you are using nightly build. Have you tried changing *output type* setting?

schwabe, 2016-06-27:
That looks the resolution of the osciloskope does not get the fast 100000 baud

projectkk2glider, 2016-06-27:
I meant the three logical levels of the signal, where one would expect only two levels.

blairfancy, 2016-06-27:
Same capture but with 2.0.19 version

[image]

schwabe, 2016-06-28:
The second one looks like 3,3V logic level while the first one is 1,8V? Is the wrong logic level only a problem with dsm2 or also with ppm/ppx/etc?

blairfancy, 2016-07-04:
As shown on peak bellow problem occurred only on DSM mode. PPM and XJT mode looks correct

[image]

blairfancy, 2016-10-18:
Hi guys,
just to confirm that bug is still present on latest 2.1 branch. What can i do to help solving this issue?

blairfancy, 2016-10-18:
is it possible to enable OUTPUT_TYPE_ROWS for dsm protocol for next release? This is probably root cause of issue. By default output type is set to push-pull output except for ppm module because g_model.moduleData[EXTERNAL_MODULE].ppmOutputType is set to 0 (open draint output type)

blairfancy, 2016-10-30:
Ok bug is solved just need to comment line 176 to 180 on file radio/src/targets/sky9x.pulses_driver.cpp
[pulses_driver.cpp.zip](https://github.com/opentx/opentx/files/560719/pulses_driver.cpp.zip)

bsongis, 2016-11-02:
@blairfancy would you open a pull request for those changes? Thanks!

blairfancy, 2016-11-02:
Sorry i m noob on git if you explain process i can do it

blairfancy, 2016-11-02:
hi i m trying to push my branch for opentx but i got access forbiden

3djc, 2016-11-02:
You need to go through a pull request

projectkk2glider, 2016-11-02:
You must make your fork this repository, push the changes there in some branch, then you create a pull request against the master branch here. Se github help for more details.

blairfancy, 2016-11-02:
ok thank Github issue workflow guide was not clear about that

kilrah, 2016-11-02:
Translator's guide has more info about the process
https://github.com/opentx/opentx/wiki/Translators-Github-Guide

Note that it can be done much more "visibly" with a GUI git client .

commit d97e4133 (2016-11-02): Fixed #3628 (#3983)

* Fixed #3628

* fixed #3628
closed 2016-11-02 by bsongis