opentx/opentx #3631: OPENTX 2.0.19 LUA BUG getLogicalSwitch for Timer1/timer2

lshems, 2016-06-26:
I know this is not a bug to be solved, since it is on OpenTX 2.0. But nonetheless, it could perhaps help people using this function to be aware of the bug:

When setting the "v1" of a LS to "Timer1" or "Timer2" , the function model.getLogicalSwitch(LS).v1 will return -59 or - 60.
The value should be positive, since negatives are reserved for the ! operator for LS and physical switches.

in OpenTX 2.1 the same will return 196 or 197.

in OpenTX 2.0 you can use 196 or 197 to set the "v1" from LUA, but it will still return -59 or -60.

kilrah, 2016-06-26:
On 2.0 v1 is a 8-bit signed int, so that's to be expected.

closed 2016-06-26 by kilrah

lshems, 2016-06-26:
Ok, good to know.