opentx/opentx #3632: openTx V2.1.8  log switch as source for input or mixer, companion ok, Taranis not as source

hrenz, 2016-06-27:
Hy,
openTx v2.1.8 
At Companion I can select a log switch (L1) as source for inputs or mixer, ok.
transfer programm to Taranis and it works well, I can see L1 as source On and OFF

At Taranis X9dPlus I can't select a log switch (L1) as source for inputs or mixers
no Log switch to selection there is possible

L1 log switch is programmed  simple as  a>x gas -90
L1 is aktiv  (gas -25)

kilrah, 2016-06-27:
Seems OK in mixes? Please confirm/detail...

[image]

r41065, 2016-06-27:
Yes its OK in Mixes,

only if you want to Use a log. switch as source in Inputs you can select them in Companion , transfer the Model to Taranis and it works.
Select log.switches as source direct in Taranis is not working.

r41065, 2016-06-27:
[image]

hrenz, 2016-06-27:
Hy,
Yes I know, at Companion ok and at Simulator ok, I can select and see L1 at mixer and at Inputs 

But not at Taranis X9DPlus with openTx V2.1.8
cant select log switches L1, L2...

kilrah, 2016-06-27:
Simu and radio should behave the same, and they do here. 

You have entered your LS before trying to choose it, right? The list hides undefined LSs.

projectkk2glider, 2016-06-27:
Confirmed

kilrah, 2016-06-27:
I've got the fix committed locally, just waiting on @hrenz to confirm mentioning the mixer is a mistake of his before pushing.

projectkk2glider, 2016-06-27:
Lua script outputs are also not available, I wonder why.

kilrah, 2016-06-27:
I think there was a reason for that, but I'm not sure anymore...

bsongis, 2016-06-27:
Yes the reason is ... nobody asked

kilrah, 2016-06-27:
I think I remember we said cascading inputs might be a bit too messy, better have things defined locally to avoid confusion, if you want to chain things you rather do it in the mixer

r41065, 2016-06-27:
I think i have to explain:
I'm building ships with Muliple Functions.
For this my biggest Model ( til now) uses 56 Mixing lines for 22 Servo-Outputs
Therefore i was searching a solution to bring a new Function to this Model using less mixing lines.
In Simulator it was well working to put a logical switch as a source to an Input Line and then use only one Mixing Line with the "Slow Motion Functinality" to build the Servo-Output ( without the Jumps in the Output if you change only the mixing lines.).

During Testing I cleared the logical switch in my Taranis and use one of the physical switches.
After that i want to put the LS back into the Input-Line but i can't select it as source again.
( Only in the Input Lines, in Mixing Lines you can select every defined LS ! )

Yesterday evening I asked the people in our German Forum if someone knews about this "Bug"
and h.renz opened this point here.

Best Regards 
Ralf

hrenz, 2016-06-27:
Well I have tested now, at Companion
I have programmed a new Modell, then I programmed 2 log Switches L1, L2,
At Companion  I can select L1, L2 and all other log switches as source of the Input, 
at the simulator I can see  at Inputs the two Log switches  all ok.
I can transmit this programm to Taranis and it works well too.
------
Then I want to progrmm the same the simulator direct.
want to select the L1 or L2 as Input, but there is No log switch to select.
[image]
[image]
[image]

see pictures

hrenz, 2016-06-27:
at mixers I can select the L1 or L2 as direct input selection, but not at the type selection menu
switches are there only phys. switches as SA,SB... but no log switches possible.
[image]
[image]

kilrah, 2016-06-27:
There has never been a separate line for logical switches in this menu, they're considered part of "switches". 
Could be an improvement, but it's unrelated.

commit f959ecea (2016-06-27): LSs weren't selectable as input source (Fix #3632)
closed 2016-06-27 by kilrah