opentx/opentx #3633: BUG - models.txt and radio.bin can be deleted from SD brower

Robomort, 2016-06-28:
It is possible to delete /RADIO/radio.bin and /RADIO/models.txt when browsing the SD card from the radio. These files should be write protected or have a warning message display when trying to delete them.

3djc, 2016-06-28:
Well, your PC also allows you to delete key elements, most of the time without warning. And when you move the sdcard to a PC, same exists. Better to learn not to delete things unless you know what you do (because their might be exceptional cases when you want to be able to do that)

Robomort, 2016-06-28:
I agree with the direct editing of the card on a PC is not something that can not be avoided but that would an extreme willful course of action and of course users will use Companion for everyday tasks when it become available. Editing these files on the radio should not be required and a simple incorrect menu selection could cause the model files to be de synced with the selection process.

kilrah, 2016-06-28:
This is irrelevant since the model storage system is going to change completely anyway.

closed 2016-06-28 by kilrah