opentx/opentx #3635: Companion 2.1.8 will not compile with gcc version 6.1.1

hursto75, 2016-06-29:
Companion 2.1.8 will not compile with gcc version 6.1.1 
Tested on fedora 24 using the below method.
As root run:
dnf install git gcc svn qt qt-devel cmake patch xsd gcc-c++ SDL-devel phonon phonon-devel avr-gcc-c++ PyQt4 fox-devel avr-libc xerces-c-devel

Run the following commands as a normal user:
git clone https://github.com/opentx/opentx
cd opentx
git checkout master
cd opentx/companion
mkdir lbuild
cd lbuild
cmake ../src
make clean
make

Hundreds of errors similar to this one.

/usr/include/c++/6.1.1/cmath:101:37: error: ‘__is_integer’ was not declared in this scope
     typename __gnu_cxx::__enable_if<__is_integer<_Tp>::__value,

May be a problem with the inclusion of the math.h lib.

projectkk2glider, 2016-06-29:
Please upload the whole compile output to pastebin and post link here.

hursto75, 2016-06-29:
Here is your link for the full build output. http://pastebin.com/AezxuH6g

projectkk2glider, 2016-06-29:
Try changing the extern "C" here https://github.com/opentx/opentx/blob/master/radio/src/lua/lua_api.h#L42 remove the ifdef and try once with exern and then without it.

projectkk2glider, 2016-06-30:
:sunglasses: your advanced compiler also found several bugs, one example in https://github.com/opentx/opentx/blob/master/companion/src/generaledit/generaledit.cpp#L158  
```
/home/bkelley6/build/opentx/companion/src/generaledit/generaledit.cpp:156:9: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
         if (ok)
         ^~
/home/bkelley6/build/opentx/companion/src/generaledit/generaledit.cpp:158:11: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
           strcpy(generalSettings.switchName[i], qba.data());
           ^~~~~~
```

projectkk2glider, 2016-06-30:
@hursto75 try with my branch #3638

hursto75, 2016-06-30:
I have tested this with the branch #3638 and it is compiling correctly. I will run thought testing of Companion tonight. Thank you.

hursto75, 2016-06-30:
When will this be branch be merged into the master?

Thanks,
Brad

projectkk2glider, 2016-06-30:
As soon as one of other devs reviews it. And don't close the issue yet, it will be closed automatically when the PR will be merged. We also need to do the same fix on the `next` branch.

hursto75, 2016-06-30:
Thanks sorry for closing.

closed 2016-06-30 by bsongis

hursto75, 2016-06-30:
Thanks again for you help. Sent 20 to the project. Keep up the excellent
work.

projectkk2glider, 2016-07-01:
I appears that `next` branch does not require any fix for gcc-6, so I only applied a part of the fix.

commit 5b517ce2 (2016-06-30): Projectkk2glider/gcc6 compilation fixes (#3638)

* Re #3635: added missing compound statements

* Fixes #3635: compilation problems with gcc-6 fixed