opentx/opentx #3637: Radiolink AT9 support?

Vimanse, 2016-06-30:
THis radio has awesome HW but a crappy FW. I took it apart and discovered it had a Stm32f103RGT6 chip so you can definitely reuse the code.Also the RF module(CC2533) is directly connected through SPI and does have a DSSS protocol.And the gimbals seems not so bad.So what do you guys think?.I you guys ready to port I can definitely do some testing.

kilrah, 2016-06-30:
That would be a large undertaking. When looking at the "system" aspect (only 3 receivers, barely anytelemtry sensor available) and the lack of functions that are pretty much a must nowadays (audio/speech) I wouldn't see the point in investing time in it when there are things around like the 9XR-Pro.

Vimanse, 2016-06-30:
But its  RF wise far superior than any tarranis.The latency is very low bcus of the DSSS and unlike tarranis which is connected through PPM ...The RF module is connected through SPI which is Ideal.Plus its not just me .Every body thinks this is a great radio(with a crappy FW).And its CHEAP.I hope you guys can reconsider

kilrah, 2016-06-30:
The Taranis' internal module isn't connected via PPM, but through a fast link that puts out a frame in 3ms. Several popular external modules (XJT, 4-in-1 multimodule, crossfire...) also use fast serial communication. 

The RF control over SPI is actually a drawback due to the reverse engineering work that would be involved to make it work and be compatible with the receivers...

Vimanse, 2016-06-30:
SO thats it theres no hope ?.should i close this?

kilrah, 2016-06-30:
There is likely no hope from the current OpenTX developers, we're already way too busy with next-gen platforms. But you can leave it open, if someone else wants to have a go they're welcome. If "everybody thinks this is a great radio" you might well find someone whi has one and has interest in porting OpenTX to it.

schwabe, 2016-06-30:
@Vimanse the main problem is support that supporting a new (undocumentated) Radio is that it takes a lot of effort. And the gains in this case seem to be minimal.

Look at DeviationTX and OpenTX, both are working on well understood hardware. Still no efforts have been to port one or the other on the other hardware platform.

davidgrm, 2016-09-26:
They have created custom firmware for this radio over at Deviation. Currently it looks like this is for use with their own radio module but it seems that the default AT-9 protocol to the TX IC is fairly simple.

davidgrm, 2016-09-26:
Is anyone else interested in porting the code to the Radiolink-AT9?
I have this radio and might make an attempt at the port... anyone interested in being involved?

Radrik5, 2017-10-24:
Could you describe what features of OpenTX will work on Radiolink AT9 after porting?

I have the transmitter and experience in programming so I could invest some time in porting but I don't fully understand what benefits I will get from it. Will the LUA scripts for PID tuning/VTX work? Does the receiver/Betaflight support it?

schwabe, 2017-10-24:
LUA will be tight. The X9D has 128 kB RAM and "only" ~70k free RAM. It depends what your goal is. If you only care about PID/VTX, reimplementing PID/VTX lua scripts in plain C/C++ in Deviation is probably less effort.

davidgrm, 2017-10-24:
Hi
Well I fly FPV and vibrations to tell you something is wrong does not help much.  I do harware design so my plan was to install a voice module. This would require software to control it. For me that would be the main thing that I wanted. Obviously have a scripting capability would be great. Perhaps something lighter than LUA?
I did make a start in writing my own firmware for the radio. I have got the screen handling and buttons sorted but unfortunatly have not had the time to work on it for a couple of months. Using a logic analyzer I also figured out all the commands between the board and the radio module

kilrah, 2017-10-29:
If you have the drivers working the biggest thing is done, integration in OpenTX shouldn't be that hard at least if screen resolution matches something we already have an UI designed for.

Narnasqueneth, 2021-01-27:
I know this is a super-old threat but I've had one of these AT9 TXs for a while and agree it could massively benefit from Opentx impelmentation. Unfortunately I have literally 0 programming skills but would be happy to volunteer as a tester. There is still interest in this radio

closed 2022-04-17 by stale[bot]