opentx/opentx #3641: OpenTX 2.1.8 countdown bug

lichtl, 2016-07-03:
Hi,

it seems when the countdown is set to beep opentx 2.1 will count down with wrong the last 10 seconds. Please see attached video captured on OpenTX 2.1.8N780 on a Taranis Plus (I was also able to replicate this on Companion).

https://www.youtube.com/watch?v=gNBllHor2BM

mhotar, 2016-07-03:
There is wrong wav file in your sound package. You can remove it, or replace with some beep sound.

mhotar, 2016-07-03:
Delete timer10.wav in SOUNDS/en/SYSTEM folder.
If you upgraded your radio from 2.0 to 2.1 recently, you should update your SD card content as well.

lichtl, 2016-07-03:
Ok. Thanks. I've used the OpenTX 2.x - Voice and Sound Pack Generator - for this.

Thanks

closed 2016-07-03 by mhotar

lichtl, 2016-07-03:
Working. Thanks alot :)