opentx/opentx #3645: Taranis stuck on startup screen

sammydigits, 2016-07-05:
My Taranis is stuck on the "Welcome to OpenTX" startup screen, and the power switch no-longer works, when i switch it to off, it stays on. None of the buttons seen to be stuck or broken. I updated the firmware which made no difference. With/without SDcard doesn't make a difference. Any ideas? Here is my [EEPROM.BIN.zip](https://github.com/opentx/opentx/files/347077/EEPROM.BIN.zip)

tomkfly, 2016-07-05:
What firmware are you using?
Have you tried removing the battery and reconnect after a couple of mins?
Did you do anything different just before the problem appeared, for example use a different charger?

schwabe, 2016-07-05:
I just cheecked the eeprom in the emulator. It does not create a hang or loop there.

kilrah, 2016-07-05:
You might want to remove battery and flash in "power off" mode.

sammydigits, 2016-07-06:
Thanks all. Aloft hobbies told me i'm still under warranty and can send it back for repair, so I'm going with that option. I appreciate your help and suggestions though.

closed 2016-07-06 by projectkk2glider