opentx/opentx #3648: Companion with proxy

francisbesset, 2016-07-05:
Hello,
I use companion behind a proxy and it is impossible to download a firmware.
I have this error: __Unable to check for updates.__

I tried to add http_proxy and https_proxy (too with caps lock) in environment variables of Windows and same error.

I don't see a parameter interface to specify the proxy in Companion.

kilrah, 2016-07-05:
What companion version?

You likely need to set the proxy in your system's network preferences (in Internet settings).

francisbesset, 2016-07-05:
I use the v2.1.8.
The proxy server is set in Internet settings but a login and password are required for proxy.

francisbesset, 2016-07-05:
Can you give me the urls called by Companion?
Maybe that proxy block theses urls.

kilrah, 2016-07-05:
http://downloads-21.open-tx.org/firmware/stamp-opentx.txt
http://downloads-21.open-tx.org/companion/companion-windows.stamp

francisbesset, 2016-07-05:
No problem to load theses urls from IE or an other browser.

closed 2019-11-15 by stale[bot]