opentx/opentx #3654: x9e 2.1.8

olimis83, 2016-07-07:
swr and alarm antenna not working in this version

tomkfly, 2016-07-07:
I replied to this on RC Groups.

projectkk2glider, 2016-07-08:
It should work. What is the actual value of SWR when you expect to see the bad antenna warning? The warning will be triggered by SWR > 51.

The things preventing SWR alarm would be:
*  a specific version of XJT module which does not report valid SWR value
* a sub-standard SWR detection circuit on your particular XJT module that does not measure SWR accurately enough.

tomkfly, 2016-07-08:
It could be he hasn't done the telemetry discovery.

kilrah, 2016-07-08:
The "antenna defective" alert works independently of any telemetry display and discovery.

tomkfly, 2016-07-08:
He said "in this version",  which made me think Thais is why he isn't getting the SWR display and he is not getting the warning because there isn't a problem.

schwabe, 2016-07-08:
Has 2.1.8 already SWR removed as telemetry only shows it under Analog inputs, or is that next only?

kilrah, 2016-07-08:
Next only.

> He said "in this version", which made me think Thais is why he isn't getting the SWR display and he is not getting the warning because there isn't a problem.

But as I said that can't be, the alarm is always active in all versions without the need for any user intervention, so if he doesn't get it there is a problem somewhere indeed.

He could help by saying whether the warning did ever work on his radio with another version or not.

free6k, 2016-07-29:
i too have this problem, on 2.1.8 i don't have alarm "antenna".
Maybe problem in my antenna?

projectkk2glider, 2017-01-14:
Most likely not an OpenTX issue.

closed 2017-01-14 by projectkk2glider