opentx/opentx #3658: fx.h simu target on 2.2

nathantsoi, 2016-07-20:
just trying to build opentx for the first time today on the next branch, i'm getting:

```
[ 97%] Building CXX object radio/src/targets/simu/CMakeFiles/simu.dir/__/__/pulses/pulses_avr.cpp.o
[ 97%] Building CXX object radio/src/targets/simu/CMakeFiles/simu.dir/simpgmspace.cpp.o
[100%] Building CXX object radio/src/targets/simu/CMakeFiles/simu.dir/__/__/simu.cpp.o
/[..redacted..]/opentx/radio/src/simu.cpp:21:10: fatal error: 'fx.h' file not found
#include "fx.h"
         ^
1 error generated.
make[3]: *** [radio/src/targets/simu/CMakeFiles/simu.dir/__/__/simu.cpp.o] Error 1
make[2]: *** [radio/src/targets/simu/CMakeFiles/simu.dir/all] Error 2
make[1]: *** [radio/src/targets/simu/CMakeFiles/simu.dir/rule] Error 2
make: *** [simu] Error 2
```

is that expected?

kilrah, 2016-07-20:
What platform?
The simu target requires libfox, that is only available on linux.

closed 2016-07-20 by kilrah

schwabe, 2016-07-20:
For reference, simu also works on OS X but currently cmake does not find fox automatically. I am using (with homebrew) the following hack. (Proper cmake integration would nice ...): 
```
diff --git a/radio/src/targets/simu/CMakeLists.txt b/radio/src/targets/simu/CMakeLists.txt
index f91c9ce..1a9d427 100644
--- a/radio/src/targets/simu/CMakeLists.txt
+++ b/radio/src/targets/simu/CMakeLists.txt
@@ -57,7 +57,7 @@ if(NOT WIN32)
   add_executable(simu WIN32 ${SIMU_SRC} ../../simu.cpp)
   add_dependencies(simu ${FIRMWARE_DEPENDENCIES})
   target_include_directories(simu PUBLIC /usr/local/include/fox-1.6 PUBLIC /usr/include/fox-1.6 /opt/local/include/fox-1.6)
-  target_link_libraries(simu FOX-1.6 pthread ${SDL_LIBRARY})
+  target_link_libraries(simu -L/usr/local/Cellar/fox/1.6.50/lib/ FOX-1.6 pthread ${SDL_LIBRARY})
 endif()```

nathantsoi, 2016-07-21:
thx for the help guys. @schwabe, `brew install fox` worked great! i'll use chat for these requests in the future.